thonny/thonny #2081: Class methods that have the same name as built-in functions don't highlight correctly.

d3jv, 2021-12-15:
Let's say i want to name a method "set". It should be higlighted in the same color as other methods because self.set() does not conflict with the set() function and set should be a legal name for a method.

aivarannamaa, 2021-12-15:
Thank you for the suggestion!

Unfortunately (performant) semantic highlighting is much harder to implement than Thonny's current regex-based approach. But for your specific case there could be a hack -- using builtins color only when the name isn't preceded by period. I'll experiment later.

d3jv, 2021-12-15:
When the name is preceded by a period, it doesn't highlight it anyway. I forgot to mention that I saw this while defining the method, so I guess maybe when it's followed by "(self". That doesn't look very clean though.
[image]