thonny/thonny #2082: Stop/Restart with WebREPL

finghra, 2021-12-15:
WebREPL connected
>>> 
MicroPython v1.17 on 2021-09-02; TinyPICO with ESP32-PICO-D4
Type "help()" for more information.
>>> 
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
ERROR   thonny.plugins.micropython.backend: Crash in backend
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/webrepl_connection.py", line 73, in _ws_connect
    self._ws = await websockets.connect(self._url, ping_interval=None)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/websockets/legacy/client.py", line 622, in __await_impl__
    transport, protocol = await self._create_connection()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 962, in create_connection
    raise exceptions[0]
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 949, in create_connection
    await self.sock_connect(sock, address)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/selector_events.py", line 473, in sock_connect
    return await fut
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/selector_events.py", line 503, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 61] Connect call failed ('192.168.1.100', 8266)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 131, in __init__
    self._process_until_initial_prompt(clean)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 202, in _process_until_initial_prompt
    self._soft_reboot_in_raw_prompt_without_running_main()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 452, in _soft_reboot_in_raw_prompt_without_running_main
    self._check_reconnect()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 518, in _check_reconnect
    self._connection = self._connection.close_and_return_new_connection()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/webrepl_connection.py", line 139, in close_and_return_new_connection
    return WebReplConnection(self._url, self._password, self.num_bytes_received)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/webrepl_connection.py", line 47, in __init__
    raise res
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/webrepl_connection.py", line 60, in _ws_main
    await self._ws_connect()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/webrepl_connection.py", line 79, in _ws_connect
    raise ConnectionFailedException(str(e))
thonny.common.ConnectionFailedException: [Errno 61] Connect call failed ('192.168.1.100', 8266)

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

aivarannamaa, 2021-12-28:
Thank you for the report!

I understand that you were able to connect to WebREPL, but after some action, you got this error. What were you doing just before getting this error? Does it happen every time?

finghra, 2021-12-29:
Yes, every time. Pressed the Stop/Restart-button immediately after WebREPL connects.

aivarannamaa, 2022-03-23:
Please see whether https://github.com/thonny/thonny/releases/tag/v4.0.0b1 works better for you

finghra, 2022-03-23:
Works now, but I still get:

`Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/workbench.py", line 1757, in event_generate
    handler(event)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/workbench.py", line 878, in _on_backend_restart
    desc = proxy.get_switcher_configuration_label(conf)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 380, in get_switcher_configuration_label
    url = conf[f"{cls.backend_name}.url"]
KeyError: 'ESP32.url'`

aivarannamaa, 2022-03-24:
> KeyError: 'ESP32.url'

Oops... 

I found the error. Will test more tomorrow. The fix will be in the next beta.

closed 2022-04-01 by aivarannamaa