thonny/thonny #2092: Thonny startup error

philippevelha, 2021-12-23:
I got the following error code twice: "maximum recursion depth exceeded while calling a Python object".
I am running on windows 10 pro. 
What can I do or check?

aivarannamaa, 2021-12-24:
Thank you for the report!

Do you only get this line? No stack-trace?

philippevelha, 2021-12-24:
I get this dialog window twice and that's it.

By the way.

Season's greetings

aivarannamaa, 2021-12-25:
Have you been able to use Thonny on this computer before? If yes, did you change anything in the settings that might have caused it?

aivarannamaa, 2022-03-23:
Please try whether https://github.com/thonny/thonny/releases/tag/v4.0.0b1 works for you

debrunner, 2022-05-06:
Get this crash randomly while scrolling source code and now also right after each start of Thonny...

On Kubuntu 18.04 with https://github.com/thonny/thonny/releases/tag/v4.0.0b2 and MicroPython 1.18 ESP32

Traceback (most recent call last):
  File "/home/td/apps/thonny/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/home/td/apps/thonny/lib/python3.10/site-packages/thonny/tktextext.py", line 1128, in _horizontal_scrollbar_update
    super()._horizontal_scrollbar_update(*args)
  File "/home/td/apps/thonny/lib/python3.10/site-packages/thonny/tktextext.py", line 843, in _horizontal_scrollbar_update
    self._hbar.set(*args)
  File "/home/td/apps/thonny/lib/python3.10/site-packages/thonny/ui_utils.py", line 980, in set
    self.grid()
  File "/home/td/apps/thonny/lib/python3.10/tkinter/__init__.py", line 2524, in grid_configure
    + self._options(cnf, kw))
  File "/home/td/apps/thonny/lib/python3.10/tkinter/__init__.py", line 1502, in _options
    cnf = _cnfmerge(cnf)
  File "/home/td/apps/thonny/lib/python3.10/tkinter/__init__.py", line 103, in _cnfmerge
    if isinstance(cnfs, dict):
RecursionError: maximum recursion depth exceeded while calling a Python object

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/td/apps/thonny/lib/python3.10/site-packages/thonny/__init__.py", line 250, in launch
    bench.mainloop()
  File "/home/td/apps/thonny/lib/python3.10/tkinter/__init__.py", line 1458, in mainloop
    self.tk.mainloop(n)
  File "/home/td/apps/thonny/lib/python3.10/tkinter/__init__.py", line 1925, in __call__
    self.widget._report_exception()
  File "/home/td/apps/thonny/lib/python3.10/tkinter/__init__.py", line 1641, in _report_exception
    root.report_callback_exception(exc, val, tb)
  File "/home/td/apps/thonny/lib/python3.10/site-packages/thonny/workbench.py", line 2388, in _on_tk_exception
    self.report_exception()
  File "/home/td/apps/thonny/lib/python3.10/site-packages/thonny/workbench.py", line 2391, in report_exception
    logger.exception(title)
  File "/home/td/apps/thonny/lib/python3.10/logging/__init__.py", line 1512, in exception
    self.error(msg, *args, exc_info=exc_info, **kwargs)
  File "/home/td/apps/thonny/lib/python3.10/logging/__init__.py", line 1505, in error
    if self.isEnabledFor(ERROR):
  File "/home/td/apps/thonny/lib/python3.10/logging/__init__.py", line 1732, in isEnabledFor
    return self._cache[level]
RecursionError: maximum recursion depth exceeded while calling a Python object

aivarannamaa, 2022-05-21:
@debrunner , thanks for the stacktrace and sorry for late reply.

I believe I have fixed this in #2258 and the fix is in 4.0.0b3.

Please reopen if you still experience this problem!

closed 2022-05-21 by aivarannamaa