thonny/thonny #2097: no attribute '_report_time'

uraich, 2021-12-28:
When starting micropython on the ESP32 I get:
Internal error ('EspMicroPythonBackend' object has no attribute '_report_time')

aivarannamaa, 2021-12-28:
Thank you for the report!

Which Thonny version are you using? How did you install Thonny? Have you used an older Thonny version before?

When you select "Tools => Manage plugins", do you see any packages starting with `thonny-`? Please also check the package named `thonny` -- where is it installed?

uraich, 2021-12-28:
I was trying to install the latest version from github. ... and yes, I was using thonny on Ubuntu for quite some time now. However, the Ubuntu version is quite old  and I need some features only available in newer versions (e.g. DTR and RTS selection).
With this old version I get a system panic after running certain tinyML code and I wanted to see if this problem disappears in the newest thonny version (it does!).
I cloned the github repository and installed thonny with:
**sudo python3 setup.py install**
This is when the problem arouse,
I finally added 
def _report_time(self,dummy):
        return
to the EspMicroPythonBackend class, which got rid of the problem and allows me to run the new thonny version. This is of course not a clean solution but worked as a work-around for me.

aivarannamaa, 2022-01-01:
Sorry, I hadn't noticed that a refactoring had broken MicroPython backends. Should be now fixed in the master.

closed 2022-01-01 by aivarannamaa