thonny/thonny #2098: No module named serial

Wojnstup, 2021-12-28:
Traceback (most recent call last):

  File "/usr/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/usr/lib/python3.10/site-packages/thonny/workbench.py", line 810, in _post_backend_menu
    entries = backend.proxy_class.get_switcher_entries()
  File "/usr/lib/python3.10/site-packages/thonny/plugins/micropython/__init__.py", line 350, in get_switcher_entries
    if cls.should_show_in_switcher():
  File "/usr/lib/python3.10/site-packages/thonny/plugins/micropython/__init__.py", line 339, in should_show_in_switcher
    potential_ports = cls._detect_potential_ports()
  File "/usr/lib/python3.10/site-packages/thonny/plugins/micropython/__init__.py", line 196, in _detect_potential_ports
    all_ports = list_serial_ports()
  File "/usr/lib/python3.10/site-packages/thonny/plugins/micropython/__init__.py", line 897, in list_serial_ports
    import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial'

I'm getting this error whenever I try to change the interpretter to Raspberry Pi Pico (Micropython). This pops up whenever I click on "Python 3.10.1" at the bottom right corner of the screen. I don't have the module "serial" installed, but I have module "pyserial" installed. I tried all the solutions I could find in the issues here, but to no effect. I'm on Arch Linux.

aivarannamaa, 2021-12-28:
Thank you for the report!

> I don't have the module "serial" installed, but I have module "pyserial" installed.

This is good. The distribution named "pyserial" is precisely what is expected to provide the module `serial`, which, for some reason, can't be found in your case.

Please tell me what do you get when executing following command in the terminal:

```
/usr/bin/python3.10 -m pip show pyserial
```

aivarannamaa, 2022-03-23:
Please reopen if you have more details to add

closed 2022-03-23 by aivarannamaa