thonny/thonny #2100: unable to connect to rpi pico

bpav3l, 2021-12-29:
in past few days i had a few problems running thonny on my linux laptop, thonny reports back:

Unable to connect to /dev/ttyACM0: [Errno 13] could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'

and then saying something about adding user to dialout group which seemed like i got either wrong or didn´t work

aivarannamaa, 2022-01-01:
Do I understand correctly, that you were able to use Thonny+Pico on this laptop before, but not anymore?

bpav3l, 2022-01-01:
sadly not, i couldnt make it work from the start

aivarannamaa, 2022-01-01:
Please do `ls -l /dev/ttyACM0` and note the group of the device. Usually it is `dialout`, but it may be something else. If it is something else, then add yourself to that group (https://www.howtogeek.com/50787/add-a-user-to-a-group-or-second-group-on-linux/). NB! Make sure you reboot after this. Several sources say logging out and in suffices, but I've seen Linuxes, where full reboot is required.

aivarannamaa, 2022-03-23:
Please reopen if you still have this problem

closed 2022-03-23 by aivarannamaa