thonny/thonny #2102: Startup issue / tkninter module

Erbie74, 2021-12-31:
Hi

Thonny gives an error at startup. My platform is Rocky Linux 8.5, and python 3.9.6. Let me paste the terminal output:

$ thonny
INFO    thonny: Thonny version: 3.3.14
Traceback (most recent call last):
  File "/usr/local/bin/thonny", line 11, in <module>
    sys.exit(launch())
  File "/usr/local/lib/python3.6/site-packages/thonny/__init__.py", line 178, in launch
    if not _check_welcome():
  File "/usr/local/lib/python3.6/site-packages/thonny/__init__.py", line 146, in _check_welcome
    from thonny.config import ConfigurationManager
  File "/usr/local/lib/python3.6/site-packages/thonny/config.py", line 9, in <module>
    import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'

aivarannamaa, 2021-12-31:
You installed Thonny with `pip`, right? Please try `sudo dnf install python3-tkinter`.

Erbie74, 2022-01-01:
Hi

Thank you for your reply.

Yes i made an install with pip.

But now i see the problem with the tkninter module. I didnt request the python3-tkninter module to install. I just typed sudo dnf install tkninter.

Erno

Hanki Outlook for Android<https://aka.ms/AAb9ysg>

aivarannamaa, 2022-01-01:
I'm glad you got it fixed!

closed 2022-01-01 by aivarannamaa