thonny/thonny #2104: Thonny in webrepl mode - shell messages not appearing.

GithubMystic, 2022-01-03:
First of all congratulations on your great Thonny app. 
It is really great and so much easier to use than other methods

I run micropython on my ESP32s
I have enabled the ESP32 webrepl
If I run thonny with my ESP32 attached via the usb port to my Windows desktop, everything works fine and the output from print('xx') statements appear in the shell
If I now disconnect the ESP32 from the usb port and power it independently and use the webrepl (under port or webrepl in run==> interpreter), everything seems to run but I no longer see messages in the shell from my print('xx') statements .
If I connect to the ESP32 via the webrepl.html web browser interface,I get the output of my print('xx') statements .
What do I need to do to be able to see the output of my print('xx') statements in the shell of Thonny whilst using the webrepl interface so that I can debug my code remotely via WiFi ?
Many thanks

GithubMystic, 2022-01-03:
I would also like to add the following in case it helps
I did find it very intermittent to actually be able to get to the list of files on teh ESP32 via the Thonny webrepl, sometimes almost giving up and then it suddenly would appear.
I tried again today so that I could perhaps find out more about getting those print('xxx') messages appearing.
Now I am no longer able to see the files on the ESP32 at all.
If I start with a running ESP32 on WiFi, I verify that I can access it via the webrepl.htlm browser interface and leave it connected
I then start Thonny in webrepl mode and I can see on the webrepl.html browser a message that says ==> concurrent WebREPL connection from ('xxx.xxx.xxx.xxx'),52182) rejected. 
If I do it a second time the 52182 becomes   52183.
I can sort of understand that as you probably can't have two parralel connections
I then disconnect from the webrepl.html browser , kill the thonny  app and restart it in webrepl mode.
I now see webREPL connected and the ESP32 is still running (I have blinking led lights to confirm the main code loop)
I now click the stop button and teh code stops running on the ESP32, but I get a lot of messages in Thonny as follows

=======> start of messages

WebREPL connected

────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
ERROR   thonny.plugins.micropython.backend: Crash in backend
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 73, in _ws_connect
    self._ws = await websockets.connect(self._url, ping_interval=None)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\legacy\client.py", line 622, in __await_impl__
    transport, protocol = await self._create_connection()
  File "C:\Program Files (x86)\Thonny\lib\asyncio\base_events.py", line 962, in create_connection
    raise exceptions[0]
  File "C:\Program Files (x86)\Thonny\lib\asyncio\base_events.py", line 949, in create_connection
    await self.sock_connect(sock, address)
  File "C:\Program Files (x86)\Thonny\lib\asyncio\selector_events.py", line 473, in sock_connect
    return await fut
  File "C:\Program Files (x86)\Thonny\lib\asyncio\selector_events.py", line 503, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 10061] Connect call failed ('192.168.0.210', 8266)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 131, in __init__
    self._process_until_initial_prompt(clean)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 202, in _process_until_initial_prompt
    self._soft_reboot_in_raw_prompt_without_running_main()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 452, in _soft_reboot_in_raw_prompt_without_running_main
    self._check_reconnect()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 518, in _check_reconnect
    self._connection = self._connection.close_and_return_new_connection()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 139, in close_and_return_new_connection
    return WebReplConnection(self._url, self._password, self.num_bytes_received)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 47, in __init__
    raise res
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 60, in _ws_main
    await self._ws_connect()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 79, in _ws_connect
    raise ConnectionFailedException(str(e))
thonny.common.ConnectionFailedException: [Errno 10061] Connect call failed ('192.168.0.210', 8266)

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

=======> end of messages

I then kill the Thony app and try to reconnect via the webrepl.html browser, but it is no longer able to connect to the webREPL on the ESP32, so Thonny has somehow killed the webREPL on the ESP32 and not just stopped the main code 
I try again to restart Thonny in webREPL and I get the following message

[Errno 10061] Connect call failed ('192.168.0.210', 8266)
Backend terminated or disconnected. Use 'Stop/Restart' to restart.

Hopefully the above information will be useful in resolving the two issues
1) Not being able to use Thonny to get to the ESP32 to maintain the code, and 
2) When I did manage to do so briefly yesterday, not being able to see the print('xxx') messages

Many thanks

GithubMystic, 2022-01-03:
Additionally, using the webREPL.html, I am able to 
-- connect
-- upload files
-- downlod files
-- restart with CTRL-D (though not always reliably, but by doing import machine then machine.reset()  to do a hard reset, the updated code on the ESP32 runs as expected, 
At least the above shows that teh ESP32 appears to be working as expected as far as webREPL functionality

GithubMystic, 2022-01-03:
A little more info that might help 
I noticed in a forum , it was mentioned that a longer sleep might be needed to enable the webrepl to break .
As I had a tight loop with  a 0.1 sleep , I tried to set it to 1 second
and tried to connect from Thonny again using webREPL and this time the CTRL-C did eventually give me the file list
Perhaps this is coincidence or the 1 second pause did the trick there.
Further tests putting the sleep back to 0.1 does work, if I do the following
==> kill and restart Thonny
==> hit Ctrl-C five or six times and wait patiently
==> the file list then appears
At least now I can then also give you a bit more debug information.
I loaded the main.py file and clicked start and got the following error messages

======> start of messages
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

WebREPL connected
>>> 
>>> 
>>> 
>>> 
>>> 
>>> %Run -c $EDITOR_CONTENT
ERROR   thonny.plugins.micropython.backend: PROBLEM WITH THONNY'S BACK-END:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 73, in _ws_connect
    self._ws = await websockets.connect(self._url, ping_interval=None)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\legacy\client.py", line 622, in __await_impl__
    transport, protocol = await self._create_connection()
  File "C:\Program Files (x86)\Thonny\lib\asyncio\base_events.py", line 962, in create_connection
    raise exceptions[0]
  File "C:\Program Files (x86)\Thonny\lib\asyncio\base_events.py", line 949, in create_connection
    await self.sock_connect(sock, address)
  File "C:\Program Files (x86)\Thonny\lib\asyncio\selector_events.py", line 473, in sock_connect
    return await fut
  File "C:\Program Files (x86)\Thonny\lib\asyncio\selector_events.py", line 503, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 10061] Connect call failed ('192.168.0.210', 8266)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 347, in _handle_normal_command
    response = handler(cmd)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 946, in _cmd_Run
    self._soft_reboot_without_running_main()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 498, in _soft_reboot_without_running_main
    self._soft_reboot_in_raw_prompt_without_running_main()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 452, in _soft_reboot_in_raw_prompt_without_running_main
    self._check_reconnect()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 518, in _check_reconnect
    self._connection = self._connection.close_and_return_new_connection()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 139, in close_and_return_new_connection
    return WebReplConnection(self._url, self._password, self.num_bytes_received)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 47, in __init__
    raise res
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 60, in _ws_main
    await self._ws_connect()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 79, in _ws_connect
    raise ConnectionFailedException(str(e))
thonny.common.ConnectionFailedException: [Errno 10061] Connect call failed ('192.168.0.210', 8266)
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 73, in _ws_connect
    self._ws = await websockets.connect(self._url, ping_interval=None)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\legacy\client.py", line 622, in __await_impl__
    transport, protocol = await self._create_connection()
  File "C:\Program Files (x86)\Thonny\lib\asyncio\base_events.py", line 962, in create_connection
    raise exceptions[0]
  File "C:\Program Files (x86)\Thonny\lib\asyncio\base_events.py", line 949, in create_connection
    await self.sock_connect(sock, address)
  File "C:\Program Files (x86)\Thonny\lib\asyncio\selector_events.py", line 473, in sock_connect
[... 78 lines cut ...]

Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 347, in _handle_normal_command
    response = handler(cmd)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 695, in _cmd_execute_source
    self._execute(source, capture_output=False)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 565, in _execute
    self._execute_with_consumer(script, self._send_output)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 698, in _execute_with_consumer
[... cut ...]
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')

======= >End of messages

So with careful navigating of procedures including killing and restarting Thonny and not using STOP/START or CTRL-D , it can sort of work, but of course I still do not get the print('xxx') message appearing in the shell

I think thatg is about as much feedback and error messages I can produce on this
Hopefully this will help a lot in resolving these issues
Many thanks

GithubMystic, 2022-01-03:
Something else I just noted. 
I added print statements in the main loop so that I could gurantee that output was produced continually.
After the ESP32 had been running a bit, I did the multiple  CTRL-C again , waited until the file list shows up, killed and restart Thonny.
The webREPL connected but it then displayed all the print messages in the shell as if it  had buffered them somewhere. ???
I repeated that process and the same happens again, the messages appear to be buffered somewhere and only show up once I restart Thonny after stopping the ESP32 with multiple CTRL-Cs

GithubMystic, 2022-01-03:
I should have also added the versions of what I am using
ESP32 is a devkit-C V2
Firmware installed on ESP32 is esp32-20210902-v1.17.bin
Thonny version is 3.3.13 (using python 3.7.9) (as  per the about tab)
Windows version is 10 (64 bit)

aivarannamaa, 2022-01-05:
Thank you for this comprehensive report!

There seem to be different issues here. The connection errors after Ctrl-D are familiar to me -- WebREPL is built on top of MicroPython VM, Ctrl-D reboots this VM and therefore it is not possible to connect right away. I probably need to make Thonny try several times before giving up. Also, when running a script (F5), Thonny 3.3 first performs soft-reboot to clear the environment. This can be turned of in the upcoming Thonny 4.0.

> Once again, the webREPL interface on the ESP32 not longer accepts connections and needs a physical restart

Do you mean it doesn't accept connections even from the browser-based client?

GithubMystic, 2022-01-05:
Hello Aivar

Thank you for your reply

In response to your question
===> Do you mean it doesn't accept connections even from the browser-based
client?
That is correct, the browser webrepl  client can no longer connect to the
ESP32 repl and the ESP32 has to be physically rebooted to get this working
again

Kind regards

aivarannamaa, 2022-03-23:
Please see whether https://github.com/thonny/thonny/releases/tag/v4.0.0b1 works better for you

HW1273, 2022-11-28:
Hello Aivar,

First of all: thanks for all you work on Thonny and especially the Micropython integration. I enjoy using the Thonny IDE as a "amateur" programmer, and I find the Micropython integration in general very convenient to use.

To me it seems that this issue is not solved yet:

- I use Thonny 4.0.1 on Windows.
- ESP01s board with 1MB memory.
- Latest Micropython version (v1.19.1)
- Started to use WEBREPL some days ago after using a USB-serial-converter before.
- I have the settings in Thonny as follows, in order not to intervene in any running script on the ESP01s' 8266.
[image]

Everything works fine, except that the output of my Micropython script (for example a small test script with a print(“Hello”) statement and time.sleep(1) in a while-loop) is not shown in Thonny. This is a script which runs automatically from boot, so for example the while-loop is in main.py which is run automatically after boot.py.

If I disconnect Webrepl in Thonny, and I connect in the “official” Micropython Webrpl client (https://github.com/micropython/webrepl), the print statement output is shown.
If I reconnect in Thonny, no script output is shown. I do see a lot of red text (“Executing…..”) and the black “Webrpl connected” text.
If I disconnect Webrepl in Thonny again (by choosing the default Python interpreter) and reconnect again in the “official” client, the Micropython script is still active and I see the print statement output again “flowing over my screen”.

If I start my script **manually** in Thonny **after connecting with Webrepl in Thonny**, the output of the print statements **is shown like it should.**
If I disconnect Webrepl in Thonny (by choosing the default Python interpreter), and I connect again, output is not shown anymore.
If I disconnect Webrepl in Thonny again, and I reconnect using the "official" Webrepl client again, script is still running and output is shown.

Any ideas? Or could I provide you with more information to assist you in further troubleshooting this in any way?

Best regards,
Henri

aivarannamaa, 2023-01-11:
@HW1273, Sorry for the late reply! Somehow I missed your comment

> If I reconnect in Thonny, no script output is shown

Did you disconnect from the web client first?

aivarannamaa, 2023-01-29:
Did you find a solution?

HW1273, 2023-01-29:
Hi Aivar,

Yes, I disconnect from the web client first. 

If I would not have, reconnect with Thonny would not succeed anyhow, because no multiple connections are allowed by the Webrepl server on the ESP-01s (you still see the connect attempt warning in the web client in that case, because you forgot to disconnect).

I work around it now, and that is OK, so it is not a real issue for me personally.

Have a nice evening (at least it is evening here now in the Netherlands)!

aivarannamaa, 2023-06-02:
Sorry for late reply!

I could now reproduce this when I turned off "Interupt working program on connect". Thonny became confused by gratuitous `>>> ` that WebREPL sent just after getting connected. 

I'll try to make it work despite of this.