thonny/thonny #2107: How to clear Thonny SSH keys

hapeldoorn, 2022-01-05:
Hello, I have exact the same problem as WGandi, but on a windows enviroment. I succeeded to secure putty acces with ssh keys & passphrase. Logging in that way via putty is no problem. But using Thonny deny's using that. Reverting to normal operations is simply not possible. I removed the SSH key option with succes in Putty and removed the content of .ssh directory at the pi. Thonny is still expecting SSH keys even when I tell it to simply use password challenge auth. A reinstall of Thonny didnt't solve the issue. Searching for known_hosts on windows and pi didn't gave 1 result / hit. Also searched the windows register for that.

I can realy need some advise to proceed, currently I can't use the pi for python programming with thonny. Hope one can help me out

aivarannamaa, 2022-01-05:
Related to #2064

hapeldoorn, 2022-01-05:
Thanks for sending me in an endless loop.

aivarannamaa, 2022-01-05:
> Thanks for sending me in an endless loop.

I didn't mean to do this :) I simply wanted to retain the link to WGandi-s report

hapeldoorn, 2022-01-05:
> > Thanks for sending me in an endless loop.
> 
> I didn't mean to do this :) I simply wanted to retain the link to WGandi-s report

Ok, no problem. Do you want me to post in #2064 to keep it combined? I coudn't do that as I started because it was closed at that moment.

aivarannamaa, 2022-01-08:
Let's continue under this issue.

hapeldoorn, 2022-01-08:
Ok, good choice. Meanwhile I found the solution for my Windows 10 laptop. Happy to share this with other users with similar issues via your GitHub. At my system the known_host file can be found at C:\Users\Henk\.ssh (Replace Henk with own username). I deleted the entry's in that file and stored it at same location. Restart Thonny and all went well.

Alle1307, 2024-05-10:
I have the same problem but with Ubuntu on my desktop.
On my desktop I also cleared the .ssh file (hidden file under home directory), was able to login via SSH but asked permission.
Tried Thonny but same problem persisted. Looks like I need to find the location of the SSHkey from Thonny but sofar not lucky.
Removing and reinstalling Thonny did not solve the problem. Looks like Thonny does not use the .ssh file but a different key file.  On my laptop (also Ubuntu) I also tried it for the first time and that worked as it should. 
It looks like I need to find the location of the SSHkey from Thonny but sofar not lucky.

nathannaidoo, 2024-10-28:
I ran into this same issue. 
-->On a windows machine open command prompt: and type: "ssh-keygen -R 192.168.x.xxx" (change to IP you are SSHing into)
this will clear the keys and now you will be able to use Thonny on Windows again.