thonny/thonny #2114: Thonny on Rpi 4 model B running Ubuntu 21.10 stuck at file saving

jasper-rpi, 2022-01-10:
Hello. I am a python beginner trying out Thonny. The problem is, that I cannot save any files. Whenever I try to choose a file destination folder, the "ok" button goes gray and I can't do anything except go through folders. The thonny application does not respond to anything and I also cant close the application. When I press "cancel" on the file destination screen, the prompt closes and opens up again. I also get a line of errors (at least I think they are errors), but i don't know if they're related to my main problem.

jasper-rpi, 2022-01-10:
I forgot to add the line of errors I get. Here it is:
```
Traceback (most recent call last):
  File "/home/jasper/.local/lib/python3.9/site-packages/thonny/__init__.py", line 183, in launch
    _delegate_to_existing_instance(sys.argv[1:])
  File "/home/jasper/.local/lib/python3.9/site-packages/thonny/__init__.py", line 278, in _delegate_to_existing_instance
    sock, secret = _create_client_socket()
  File "/home/jasper/.local/lib/python3.9/site-packages/thonny/__init__.py", line 321, in _create_client_socket
    client_socket.connect(get_ipc_file_path())
ConnectionRefusedError: [Errno 111] Connection refused
`````

aivarannamaa, 2022-01-11:
Thank you for the report!

This kind of error output appears after Thonny has been closed unsuccessfully, so most likely this isn't the reason Thonny misbehaves.

Have you been able to use Thonny properly on this machine and distro version before?

How did you install Thonny? Which version?

aivarannamaa, 2022-01-11:
> the prompt closes and opens up again

Does this happen both with Tk and Zenity dialogs?

closed 2022-01-11 by jasper-rpi