thonny/thonny #2116: Unable to connect to COM5: port not found  Backend terminated or disconnected. Use 'Stop/Restart' to restart.

ocdarragh, 2022-01-13:
I keep getting this message on Thonny....any ideas?
It seems to be constantly running. When I hit stop it happens again.
I cannot run any file

aivarannamaa, 2022-01-13:
Do you want to run MicroPython or Python 3 code?

ocdarragh, 2022-01-14:
Python 3 code.

I was trying to connect a microbit through Thonny but now I am trying to run a simple Python file

Thanks,
Darragh

Sutton Park School is a Designated Activity Company (DAC) with registered offices at St Fintan's Road, Sutton, Dublin 13 Incorporated in Dublin, Ireland. Company Registration Number: 16373
Sutton Park School is a Designated Activity Company (DAC) with registered offices at St Fintan's Road, Sutton, Dublin 13 Incorporated in Dublin, Ireland. Company Registration Number: 16373

aivarannamaa, 2022-01-14:
In this case you need to switch back to Python3 mode. Click the button on the lower_right corner of the main window.

ocdarragh, 2022-01-14:
That worked!
Thanks

closed 2022-01-17 by aivarannamaa