thonny/thonny #2117: 'ShellView' object has no attribute 'pointer'

TomEpsilon, 2022-01-14:
Good morning. 
Today, I tried to install Thonny. But when I open the program, I get errors. Here's the first one:
`Traceback (most recent call last):
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1470, in _show_views
    self.show_view(view_id, False)
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1647, in show_view
    view = self.get_view(view_id)
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1553, in get_view
    view = class_(self)  # View's master is workbench to allow making it maximized
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 119, in __init__
    undo=True,
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1494, in __init__
    super().__init__(master, view, cnf=cnf, **kw)
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 311, in __init__
    super().__init__(master, cnf, **kw)
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\ui_utils.py", line 818, in __init__
    master=master, style=style, tag_current_line=tag_current_line, cnf=cnf, **kw
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\codeview.py", line 36, in __init__
    super().__init__(master=master, cnf=cnf, **kw)
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\tktextext.py", line 207, in __init__
    self._reload_theme_options()
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\codeview.py", line 70, in _reload_theme_options
    self._reload_syntax_options(event)
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\codeview.py", line 74, in _reload_syntax_options
    self.set_syntax_options(_syntax_options)
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1478, in set_syntax_options
    self.update_margin_color()
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1482, in update_margin_color
    self.tag_configure("io", lmargincolor=get_syntax_options_for_tag("TEXT")["background"])
KeyError: 'background'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 246, in set_scrollbar
    self.update_plotter()
  File "C:\Users\hieut\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 263, in update_plotter
    if self.plotter is not None and self.plotter.winfo_ismapped():
AttributeError: 'ShellView' object has no attribute 'plotter'`

Apparently there is an update in a module. I suggest an update to fix this.

TomEpsilon, 2022-01-14:
Update: Here is a screenshot of what I see when launching Thonny. The errors are dragged for clearer view.
[image]

TomEpsilon, 2022-01-14:
I also tried to uninstall and reinstall Thonny multiple times, to no avail.

TomEpsilon, 2022-01-14:
Here is the `configuration.ini` file:
```[general]
configuration_creation_timestamp = 2022-01-10T20:07:18.398731
language = en_US
environment = []
single_instance = True
event_logging = False
disable_notification_sound = False
debug_mode = False
ui_mode = regular
scaling = default
font_scaling_mode = default

[view]
full_screen = False
maximize_view = False
shellview.visible = True
ui_theme = Raspberry Pi Dark
syntax_theme = One Dark
editor_font_size = 13
editor_font_family = Consolas
io_font_size = 11
io_font_family = Courier New
assistantview.visible = False
astview.visible = False
stackview.visible = False
exceptionview.visible = False
filesview.visible = False
heapview.visible = False
helpview.visible = False
notesview.visible = False
objectinspector.visible = False
outlineview.visible = False
variablesview.visible = False
show_program_arguments = False
show_plotter = False
name_highlighting = False
locals_highlighting = False
paren_highlighting = True
syntax_coloring = True
highlight_tabs = True
highlight_current_line = False
show_line_numbers = True
recommended_line_length = 0

[file]
recent_files = ['D:\\Tom\\work\\sh\\wooper\\__init__.py']
current_file = D:\Tom\work\sh\wooper\__init__.py
open_files = ['D:\\Tom\\work\\sh\\wooper\\__init__.py']
reopen_all_files = False

[run]
backend_name = CustomCPython
working_directory = D:\Tom\work\sh\wooper
program_arguments = 
dock_user_windows = False
pgzero_mode = False
auto_cd = True
birdseye_port = 7777
run_in_terminal_python_repl = False
run_in_terminal_keep_open = True

[assistance]
disabled_checks = []
open_assistant_on_errors = True
open_assistant_on_warnings = False
use_pylint = True
use_mypy = True

[CustomInterpreter]
path = C:\Users\hieut\AppData\Local\Programs\Python\Python310\python.exe
used_paths = ['C:\\Users\\hieut\\AppData\\Local\\Programs\\Python\\Python310\\python.exe']

[layout]
zoomed = True
notebook_nw_visible_view = None
notebook_w_visible_view = None
notebook_sw_visible_view = None
notebook_s_visible_view = ShellView
notebook_ne_visible_view = None
notebook_e_visible_view = None
notebook_se_visible_view = None
west_pw_width = 299
east_pw_width = 299
nw_nb_height = 299
sw_nb_height = 299
s_nb_height = 299
se_nb_height = 299
ne_nb_height = 299

[edit]
tab_complete_in_editor = True
tab_complete_in_shell = True
indent_with_tabs = False

[debugger]
frames_in_separate_windows = True
automatic_stack_view = True
allow_stepping_into_libraries = False
preferred_debugger = nicer

[shell]
tty_mode = True
max_lines = 1000
squeeze_threshold = 1000
auto_inspect_values = True
```

aivarannamaa, 2022-01-15:
Thank you for the report!

From the configuration file I see that you're using a 3rd party syntax theme. Can you switch to a built-in syntax theme and try again?

aivarannamaa, 2022-01-15:
I think I got it. I was able to receive similar error when I uninstalled the 3rd party theme, without switching to another theme first.

TomEpsilon, 2022-01-15:
Sure! Let me try.

TomEpsilon, 2022-01-15:
Yep! It worked.

aivarannamaa, 2022-01-16:
Fix will be in 4.0b1

closed 2022-01-16 by aivarannamaa