thonny/thonny #2118: Thonny creates line numbers too long

smblackledge, 2022-01-14:
When creating pages to print, if the line count is over 999, the line count column overflows into the next line. This causes the code lines to be double spaced. Either an option to turn off line numbering needs to exist or the line number column needs to be wider if there are more than 999 lines to be printed in the listing.

aivarannamaa, 2022-01-15:
Thank you for the report!

Which Thonny version are you using?

smblackledge, 2022-01-19:
3.3.10 (what I got from "apt update" on the Raspberry Pi).

aivarannamaa, 2022-01-20:
I was hoping that you had older version so that I could say that the problem was fixed in #1647 :)

To my surprise, the problem appears in the latest version (and is actually different from #1647). Not sure, whether the browser have changed or I didn't test thoroughly enough last time.

I implemented the fix, which will be published starting with Thonny 4.0.0b1

closed 2022-01-20 by aivarannamaa