thonny/thonny #2122: celluloid not recognized

f5exo, 2022-01-15:
ModuleNotFoundError: No module named 'celluloid'

from celluloid import Camera

The script run good in linux terminal, but not with thonny.

if you have any solution .

Thank's

Erick

aivarannamaa, 2022-01-15:
If you installed Thonny+Python bundle, then the default interpreter does not use system's Python packages. In this case you either need to install `celluloid` for Thonny's Python as well (Tools => Manage packages) or make Thonny use system Python for running your programs (the button in the lower-right corner of the main window).

f5exo, 2022-01-24:
Thank you for responding to my request for help.
Indeed, I was able to install the celluloid package from the tool manage package menu.
And so I can run the script without any problem.
A big thank you to you.

Erick

aivarannamaa, 2022-01-24:
You're welcome!

closed 2022-01-24 by aivarannamaa