thonny/thonny #2123: Thonny is not usable in MAC

Cod3rror, 2022-01-15:
Hello,

Good day!

I cannot paste, debug, or run the code when I open the app. I don't know what happened, and yesterday it was ok.

I have already uninstalled and reinstalled it, and I also tried to restart the mac. Same result I get.

Thank you for helping me in advance.

[image]

aivarannamaa, 2022-01-15:
Thank you for the report!

> I cannot paste, debug, or run the code when I open the app

What does it mean precisely? Do you get an error message, are the commands disabled or just nothing happens when you press the button, select menu-item or enter keyboard shortcut?

Does the problem happen with this script only? From the screenshot it looks like syntax coloring isn't working properly. Can you zip the file as it is and post it under this issue?

aivarannamaa, 2022-03-23:
Please reopen if you have more information to add

Cod3rror, 2022-03-23:
> Thank you for the report!
>
>
> What does it mean precisely? Do you get an error message, are the commands disabled or just nothing happens when you press the button, select menu-item or enter keyboard [...]

I am sorry for replying just now.

I have no error message. Just nothing happen when I press button(i cannot press button). Also I cannot paste any script.

let me try again first, it might no longer issue now. Actually, thonny helps me alot to understand the code when I am learning python.

aivarannamaa, 2022-03-23:
If you still have this problem then you could try the first beta of 4.0: https://github.com/thonny/thonny/releases

Cod3rror, 2022-03-23:
Ok thanks. Ill check it when I got home.

Cod3rror, 2022-03-23:
> If you still have this problem then you could try the first beta of 4.0: https://github.com/thonny/thonny/releases

I think the code I tried to test when I am a beginner is the only one that makes the Thonny hang. I tried my new and more complex codes and all was working. you could check this line of code and try it. it might be a bug.

#Write your code below this line 👇
def prime_checker(number):
    if number == 2:
        print("It's a prime number.")
    elif number % 2 >= 1:
        print(f"It's a prime number.")
    elif number % 3 >= 1:
        print(f"It's a prime number.")
    else:
        print("It's not a prime number.")

#Write your code above this line 👆
    
#Do NOT change any of the code below👇
n = int(input("Check this number: "))
prime_checker(number=n)

Cod3rror, 2022-03-23:
I found the bug.. the emoticons.

aivarannamaa, 2022-05-13:
The 4.0 branch will use the newest Tk, and this is the best I can do about emoticon support.

closed 2022-05-13 by aivarannamaa