thonny/thonny #2126: [bug] about box dissapears on mouse over of menu icons

SpudGunMan, 2022-01-17:
osx12 8.6.8 

mouseover/tooltips or whatever also closes the about box 

it was a bit of a ' cat and mouse ' game to catch the window to read about it all!

SpudGunMan, 2022-01-17:
interesting the app also now shows a different about box and version 3.3.13

SpudGunMan, 2022-01-17:
little more on this as I see a change to the credits in the last 14 hours

new install, the about box was working fine and something crashed either the about box or the app or the preferences tab and then once that happened I saw the new 3.3.13 about box grey and such (the old one was simpler looking with a scrolling textbox in the middle)

aivarannamaa, 2022-01-18:
I didn't quite get what's the problem. Can you post a screenshot?

SpudGunMan, 2022-01-18:
[image]

aivarannamaa, 2022-03-23:
Please check, whether https://github.com/thonny/thonny/releases/tag/v4.0.0b1 works better for you