thonny/thonny #2127: [bug] 'preferences' screen unbound "tab" causing fault

SpudGunMan, 2022-01-17:
includes an additional tab that was not intended for production release. unbound and causes a fault also unbound. could inject code here etc. 

osx12 3.1.13

reproduce by opening preferences and clicking the "untitled tab", crash

aivarannamaa, 2022-01-18:
Thanks for the report! I can't reproduce it, though. Can you post a screenshot?

SpudGunMan, 2022-01-18:
also its not a fault I misspoke the windows seem to fly around when clicked

[image]

aivarannamaa, 2022-01-18:
I see. It looks like the same problem, which is mentioned for IDLE in https://bugs.python.org/issue34864 and https://github.com/python/cpython/blob/f4c03484da59049eb62a9bf7777b963e2267d187/Lib/idlelib/macosx.py#L115

            'WARNING: The system preference "Prefer tabs when opening'
            ' documents" is set to "Always". This will cause various problems'
            ' with IDLE. For the best experience, change this setting when'
            ' running IDLE (via System Preferences -> Dock).'

Does changing this preference fix Thonny for you?

aivarannamaa, 2022-01-18:
Probably related to #1721

SpudGunMan, 2022-01-18:
check this out https://www.reddit.com/r/mac/comments/qp6t25/monterey_ignoring_prefer_tabs_always_when_opening/

anyway looks like its a mess, just never seen it till now!

aivarannamaa, 2022-03-23:
Did you find a solution?

closed 2023-01-29 by aivarannamaa