thonny/thonny #2131: No easy way to remove colorizing name

smblackledge, 2022-01-19:
If I select Find and then use "grid" for my search target, "grid" is highlighted in red.  Now, try to get of the highlighting.  I haven't looked hard, but I have looked.  Haven't found the way to do it yet.

aivarannamaa, 2022-01-20:
Do I understand properly, that closing the Find dialog doesn't remove the highlight?

smblackledge, 2022-01-21:
Closing the Find dialog does remove the highlight.  I hadn't thought of doing that. On the other hand, we may have a difference of opinion on how the highlighting feature should work.

Looking at the problem again, I think I can make a case for having the two actions (closing the dialog and removing the highlight) be unrelated. For example, if I want space for viewing, I should be able to get rid of the dialog and, at the same time, keep the highlights for further reference? I'm still debugging, but I just don't need the dialog at the moment.

The reverse is also true. If I want to keep the dialog handy because I'm debugging. Imagine that I've finished with one problem, I want to remove highlighting, do some work unrelated to the former "find", and then enter my next find?

aivarannamaa, 2022-03-23:
Sorry for the late reply!

The old highlighting is also removed when you search for the new string. Therefore I don't see that another UI element or command for removing the highlighting, is justified here. 

I'm closing this in favor of #117, which would add another way for removing the highlight -- deleting the search string

closed 2022-03-23 by aivarannamaa