thonny/thonny #2134: feature request binary encoding type to hex

SpudGunMan, 2022-01-21:
adding binary as an option to open for loading and saving would be nice to move support files on and off connected devices and check readable assets in the binary

https://github.com/thonny/thonny/blob/a338f2892d10e9a9e50b4b4417ffd69d2d1837a5/thonny/editors.py#L198

this section here appears to look for and deny binary.. could something be added to read and write them into hex?

```
 >>> import binascii
 result = (binascii.hexlify(filename))
```

aivarannamaa, 2022-01-21:
You can transfer any files via Files view (View => Files). Right-click on a file on the local (or device) pane and select "Upload ..." (or "Download ...")

SpudGunMan, 2022-01-21:
oh snap, I feel like a fool. very nice.

aivarannamaa, 2022-01-21:
You're not the first one who failed to discover this feature. I'll see whether I can do something about his.

closed 2022-01-21 by aivarannamaa