thonny/thonny #2136: Thonny start crashing when try to type in the shell

amir-200, 2022-01-22:
I want to use thonny for programming my raspberry Pi pico but I am not able to do anything all the functions are also not showing up(run, debug,etc) please help

[image]

aivarannamaa, 2022-01-22:
Thank you for the report!

It is unfortunate that Thonny gives this error dialog instead of simply refusing accepting input, but you still need to deal with the message in red before you can start using your Pico.

amir-200, 2022-01-23:
Thanks for your compliment,
But how can I achieve that??

aivarannamaa, 2022-01-23:
> But how can I achieve that??

You mean adding yourserf to `dialout` group? As the message says ...
For example, if your username is `amir`, then execute following in the Terminal:

`sudo usermod -a -G dialout amir`