thonny/thonny #2139: Style suggestion: "NB" is archaic and should not be used.

jharris1993, 2022-01-24:
Greetings!

I have noticed that the authors periodically use "nb" to indicate a comment or notation.

Stylistic suggestion:
Using "NB" to indicate an editorial comment is somewhat archaic, overly formal, and should be avoided.

As a general rule, most comments should be specified without any notation marks.&nbsp; If a comments needs to be emphasized, it can be placed within a comment-box, highlighted, or made a notation.

"NB", when used, is **always capitalized**.

Examples:
1.  On the "Deployment Options" page, there is the note:
> nb! These instructions apply to Thonny since version 3.0

- "NB", when used, is **always capitalized**.
- It's better to simply avoid using it as sown below.

> These instructions apply to Thonny since version 3.0

2.  If something needs particularly strong emphasis, a comment-box can be used as is shown on the FAQ page located at [https://github.com/thonny/thonny/wiki/FAQ#i-want-to-help-by-translating-thonny-ui-or-webpage-to-another-language-how-should-i-start](https://github.com/thonny/thonny/wiki/FAQ#i-want-to-help-by-translating-thonny-ui-or-webpage-to-another-language-how-should-i-start)

3.  If particular emphasis is desired within a paragraph or an article, you can "notate" the comment like this:
>Note that this only affects versions of Thonny prior to version 2.7.8.

However, it is usually more effective to change the typeface or weight to emphasize an important note.
>***This only affects versions of Thonny prior to version 2.7.8.***

Thank you for an excellent product.&nbsp; Please let me know if I can do anything else to help.

aivarannamaa, 2022-01-31:
Thank you for the suggestion!

It's really funny -- I used to write capitalized "NB!" in Thonny's wiki (as I do in my mother language) until someone came and convinced me it should be lowercase :)

At the moment I'm rather busy for preparing the first beta of 4.0, but I'll deal with the nb-s after this.

jharris1993, 2022-01-31:
No problem. Do what you can, when you can.

Style can be a vexing problem which depends on what  "version" of a language you're using.

American English is stylistically different from British English. (And don't even get me starteed about Australian English!)

My granddaughters are learning "English" in their Russian schools and some of the things they are being taught is "normal" and correct would be considered atrocious grammar in American English. (Using "got" as a drop-in replacement for "have", or "yummy" instead of "delicious".)

It makes editors crazy!

KubaO, 2023-05-18:
I imagine NB is more of a European "thing". I would not consider it necessarily a stylistic issue in vacuo. Rather, it doesn't belong in Thonny materials such as documentation or wiki, because it is complexity for complexity's sake. Instead of *NB* just write *Note:*, or *Please note:*. Same meaning, less obfuscation in a project targeting a worldwide audience. The majority of Thonny's userbase has English as a 2nd language I'd think?