thonny/thonny #2140: Thonny and Waveshare ESP8266 / Pi Pico

colinhiscock, 2022-01-25:
Not sure if this is a Thonny or MicroPython thing but I have a wave share ESP8266 unit connected to my Pico and it works just fine when connected to my Mac on Thonny.  However, when on an external power source (yes, I've checked the voltage) it stalls when connecting to wifi.   I've done some basic fault finding using the LED's and they stop blinking when it gets to the line: 'wifi_connect(SSID,PASSWORD)' so I suspect it's something to do with the ESP8266 unit? I've read somewhere that there's a thing where Thonny executes the code differently to how it's run on the Pico itself but not sure how to investigate this

aivarannamaa, 2022-01-31:
Thank you for the report!

Unfortunately I didn't quite understand your setup. Have you tried doing the same without Thonny?

colinhiscock, 2022-01-31:
Aivar,

Thanks for your reply.

Yes, the issue is that it doesn’t work when it’s not plugged into Thonny.  It’s fine when it’s connected to Thonny but when it’s on external power (even from a plug) then it doesn’t work. It starts up but nothing happens. I was wondering if it’s a things in the method that Thonny writes the data to the Pico?

Colin

Regards,
Colin Hiscock
Technical Director
01243 787150
***@***.***
yourenvironment.org
This message is private and confidential.  If you have received this message in error, please notify us and remove it from your system.  Your Environment Ltd may monitor email traffic data and also the content of email for the purposes of security and staff training.  Your Environmental Company Ltd, trading as Your Environment, registered in England and Wales No:  07717486.  Registered address: Unit 2, Woodhorn Business Centre, Oving, Chichester, PO20 2BX.

aivarannamaa, 2022-01-31:
Oh, in this case -- how did you save your script on the device? As main.py?

colinhiscock, 2022-01-31:
Aivar,

Yes I did so I’m not sure what’s gone wrong. Sorry!

Col

Regards,
Colin Hiscock
Technical Director
01243 787150
***@***.***
yourenvironment.org
This message is private and confidential.  If you have received this message in error, please notify us and remove it from your system.  Your Environment Ltd may monitor email traffic data and also the content of email for the purposes of security and staff training.  Your Environmental Company Ltd, trading as Your Environment, registered in England and Wales No:  07717486.  Registered address: Unit 2, Woodhorn Business Centre, Oving, Chichester, PO20 2BX.

aivarannamaa, 2022-01-31:
In this case I suggest you ask in MicroPython forum.

closed 2022-03-23 by aivarannamaa