thonny/thonny #2142: internal error

romeovictor, 2022-01-27:
good day,
i often has this error while loading a file from the esp connected with repl (but not for all files)

Traceback (most recent call last):
  File "P:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "P:\Program Files (x86)\Thonny\lib\site-packages\thonny\base_file_browser.py", line 638, in <lambda>
    label=tr("Open in Thonny"), command=lambda: self.open_file(selected_path)
  File "P:\Program Files (x86)\Thonny\lib\site-packages\thonny\base_file_browser.py", line 1005, in open_file
    get_workbench().get_editor_notebook().show_remote_file(path)
  File "P:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py", line 952, in show_remote_file
    return self.show_file(make_remote_path(target_filename))
  File "P:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py", line 939, in show_file
    self.select(editor)
  File "P:\Program Files (x86)\Thonny\lib\tkinter\ttk.py", line 892, in select
    return self.tk.call(self._w, "select", tab_id)
_tkinter.TclError: .!frame.!automaticpanedwindow.!automaticpanedwindow2.!editornotebook.!editor5 is not managed by .!frame.!automaticpanedwindow.!automaticpanedwindow2.!editornotebook

aivarannamaa, 2022-02-28:
Sorry for the late reply! Your message somehow escaped me. I'll make a code review and try to reproduce this error.

aivarannamaa, 2022-03-27:
Unofrtunately I haven't found anything suspicious and have no idea how to reproduce this.

Can you give me some clues? For example, does it happen more ofte in the beginning of the session? More often with larger or smaller files? When it's the first file to be opened or there are several files open. 

From the screenshot I gather that you got this error while right-clicking an item in the Files view and selecting "Open in Thonny", Have you received this error while opening the file by double-clicking?

If you experience this again, then please send me the screenshot of the whole Thonny's window.

abulka, 2023-04-07:
I'm getting this error when I double click on `main.py` on my Raspberry Pico.  It never used to happen. And that's a critical file - so am stuck. Thony opens other files called `main.py` on the local pc ok. And Thonly opens other files ok.

```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/base_file_browser.py", line 586, in on_double_click
    self.open_file(path)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/base_file_browser.py", line 1201, in open_file
    get_workbench().get_editor_notebook().show_remote_file(path)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/editors.py", line 1069, in show_remote_file
    return self.show_file(make_remote_path(target_filename))
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/editors.py", line 1056, in show_file
    self.select(editor)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/tkinter/ttk.py", line 885, in select
    return self.tk.call(self._w, "select", tab_id)
_tkinter.TclError: .!frame.!automaticpanedwindow.!automaticpanedwindow2.!editornotebook.!editor2 is not managed by .!frame.!automaticpanedwindow.!automaticpanedwindow2.!editornotebook
```

I did just install `brew` and the mac command line tools, not sure if that is relevant.

abulka, 2023-04-07:
Now I'm getting an intermittent
```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/workbench.py", line 1106, in dispatch_from_menu
    dispatch(None)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/workbench.py", line 1032, in dispatch
    handler()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/running.py", line 483, in cmd_run_current_script
    self.execute_current("Run")
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/running.py", line 415, in execute_current
    filename = editor.save_file()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/editors.py", line 308, in save_file
    result = self.write_remote_file(save_filename, content_bytes, save_copy)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/editors.py", line 377, in write_remote_file
    result = get_runner().send_command_and_wait(
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/running.py", line 293, in send_command_and_wait
    show_dialog(dlg)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/ui_utils.py", line 2222, in show_dialog
    dlg.wait_visibility()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/tkinter/__init__.py", line 730, in wait_visibility
    self.tk.call('tkwait', 'visibility', window._w)
_tkinter.TclError: window ".!inlinecommanddialog9" was deleted before its visibility changed
```

Thony `4.0.2`.

abulka, 2023-04-09:
Also now cannot open `main.y` in another subdir of my pico.

And now, it gets worse... cannot delete `main.py` on my pico root directory.
```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/base_file_browser.py", line 838, in delete
    self.perform_delete(selection["paths"], tr("Deleting %s") % selection["description"])
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/base_file_browser.py", line 1252, in perform_delete
    get_runner().send_command_and_wait(
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/running.py", line 293, in send_command_and_wait
    show_dialog(dlg)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/ui_utils.py", line 2231, in show_dialog
    dlg.focus_set()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/tkinter/__init__.py", line 765, in focus_set
    self.tk.call('focus', self._w)
_tkinter.TclError: bad window path name ".!inlinecommanddialog15"
```

Please help!

[image]

tried again and got

[image]

rebooted the pico and the file was gone ok.

Opening the main.py in the subdir:

[image]

Is the name `main.py` stored somewhere in Thony for some reason, in a preference or cache that might be doing this? Can I clear things somehow?

aivarannamaa, 2023-04-09:
@abulka, thanks for the additional info!

Unfortunately I still don't know how to reproduce this. I suspect the error happens when there are some problems with communicating with the device. Therefore, if you experience this again and backend.log is mentioned, then select "Tools => Open Thonny data folder" and send me the content of backend.log. This needs to be done right after you get the error. If you restart the connection, the old file gets overwritten.

abulka, 2023-04-09:
Ok here are the log files. The error this morning happens to be another file on the pico I can't open, so its not just main.py that has the problem. Overall Thonny has been working ok except for these glitches.
[backend.log](https://github.com/thonny/thonny/files/11186520/backend.log)
[frontend.log](https://github.com/thonny/thonny/files/11186521/frontend.log)