thonny/thonny #2145: Management

uliseswolo, 2022-01-30:


uliseswolo, 2022-01-30:
can´t manage to install any package, keep showing that error or [Errno 2] ENOENT (missing file as I looked it up)

aivarannamaa, 2022-01-31:
Thank you for the report!

Do you have a `/lib` directory next to your `main.py` (ie. `/lib` or `/flash/lib`, depending on the device)?  If not then create it manually

[image]

and try again.

If this helps, then please let me know!

aivarannamaa, 2022-01-31:
The right folder name can be queried with:

```
import sys
print(sys.path)
```

aivarannamaa, 2022-03-23:
Were you able to make it work after creating the directory manually? If not then please try https://github.com/thonny/thonny/releases/tag/v4.0.0b1

Small-Wooden-Soul, 2022-04-20:
Hi, @aivarannamaa .  Firse, please forgive me my poor English. 
I have the similar problem.
I want to install **_micropython-mpu9250_** package. But I can't install package by Manage packages.

[image]

The error of minipip showed is same as @uliseswolo  .

I saw you said create `/lib` if we don't have this directory. 
But I'm not sure were you meant this?

[image]

I thought I already have `/lib` directory. But I still can't install successful.
Am I misunderstand what you mean? Or I just missing something?

closed 2023-01-28 by aivarannamaa