thonny/thonny #2146: Plug-in manager broken?

CrackXT, 2022-01-30:
Last officially linked downloadable version of Thonny on the website -> 3.3.13

Thonny -> Tools -> Mangae plug-ins...

See Example, no buttons are displayed ?
[image]
With kind regards
CrackXT

WMRamadan, 2022-01-30:
From the screenshot is it fair to assume you are running this on Windows 11?

aivarannamaa, 2022-01-31:
With this dialog you can upgrade or uninstall only these packages which you installed yourself.

In this case I now that Thonny accepts newer version as well, so if you need newer esptool, then select "The same interpreter which runs Thonny" backend (from the lower-right corner of Thonny's main window), then select "Tools => open System shell" and enter `pip install -U esptool`.

Thonny 4.0 will comes with newer esptool.

CrackXT, 2022-01-31:
@ WM Ramadan Thanks for your feedback. Yes is a Windows 11 system.

@aivarannamaa Thanks for your feedback. Oh, I thought that refers directly to Thonny because the specified path is also the one from the Thonny folder.

Since I already had everything up to date under C:\WINDOWS\system32.

Does that mean if I update it via CMD (Tools => open system shell) then Thonny uses the versions in the newly linked path?

With kind regards
CrackXT

aivarannamaa, 2022-03-23:
Sorry for the late reply!

> Does that mean if I update it via CMD (Tools => open system shell) then Thonny uses the versions in the newly linked path?

Not quite. It simply replaces the old esptool. But as alternative, consider trying https://github.com/thonny/thonny/releases/tag/v4.0.0b1 which comes with newest esptool

closed 2023-01-29 by aivarannamaa