thonny/thonny #2147: Issue installing 3.3.7-x86_64-alt on Linux

villares, 2022-01-30:
Context: I'd like to use this version because of the Python version bundled with it (in order to use py5.ixora.io).  As a workaround I've been able to work inside an env I created with conda, after installing Thonny with pip on it, but it is not ideal, as I have to activate the env etc.).  On Windows (for my students) it seems to install fine.

Installing the thonny-3.3.7-x86_64-alt.tar.gz with the provided install.py on Manjaro-XFCE I get these errors:

```
Traceback (most recent call last):
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/workbench.py", line 1310, in get_settings
    parent, settings = self._syntax_themes[name]
KeyError: 'Kyanite Syntax'
```
and then 
```
Traceback (most recent call last):
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/workbench.py", line 1469, in _show_views
    self.show_view(view_id, False)
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/workbench.py", line 1646, in show_view
    view = self.get_view(view_id)
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/workbench.py", line 1552, in get_view
    view = class_(self)  # View's master is workbench to allow making it maximized
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/shell.py", line 105, in __init__
    self.text = ShellText(
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/shell.py", line 1494, in __init__
    super().__init__(master, view, cnf=cnf, **kw)
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/shell.py", line 311, in __init__
    super().__init__(master, cnf, **kw)
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/ui_utils.py", line 817, in __init__
    super().__init__(
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/codeview.py", line 36, in __init__
    super().__init__(master=master, cnf=cnf, **kw)
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/tktextext.py", line 207, in __init__
    self._reload_theme_options()
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/codeview.py", line 70, in _reload_theme_options
    self._reload_syntax_options(event)
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/codeview.py", line 74, in _reload_syntax_options
    self.set_syntax_options(_syntax_options)
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/shell.py", line 1478, in set_syntax_options
    self.update_margin_color()
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/shell.py", line 1482, in update_margin_color
    self.tag_configure("io", lmargincolor=get_syntax_options_for_tag("TEXT")["background"])
KeyError: 'background'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/villares/apps/thonny/lib/python3.9/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/shell.py", line 246, in set_scrollbar
    self.update_plotter()
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/shell.py", line 263, in update_plotter
    if self.plotter is not None and self.plotter.winfo_ismapped():
AttributeError: 'ShellView' object has no attribute 'plotter'
```
Some more messages and the editor closes...

tabreturn, 2022-01-30:
```
Traceback (most recent call last):
  File "/home/villares/apps/thonny/lib/python3.9/site-packages/thonny/workbench.py", line 1310, in get_settings
    parent, settings = self._syntax_themes[name]
KeyError: 'Kyanite Syntax'
```
This is a Thonny config issue. The py5mode plugin (that includes the Kyanite theme) isn't installed, so Thonny cannot load that theme. Delete the Thonny config file (or the offending line in the config).

tabreturn, 2022-01-30:
On this point -- 

Thonny should probably include some workaround to load the default theme if a configured one is missing (i.e. if/when the user uninstalls/deletes a theme plugin).

villares, 2022-01-31:
Thank you @tabreturn! I suppose I'll close this now!

I couldn't find the config files, but I changed themes in the version running from the conda env and deleted a few thonny named folders all around :( 
Reinstalled and it's working... but I have to point it to the conda env environment interpreter!!!

I noticed that running Thonny from the conda env made the UI behave terribly strange, fonts, window sizes, now it looks much better... that was another reason for me trying to install it again. Now I can launch from the desktop icon or Linux search, much, much better.

closed 2022-01-31 by villares

tabreturn, 2022-01-31:
Just for future reference --

The solution is to delete the offending theme line(s) from your *configuration.ini* file, located:

- Linux config path: `~/.config/Thonny`
- Windows config path: `\Users\<user>\AppData\Roaming\Thonny`
- macOS config path: `~/Library/Thonny`

Logged the 'theme-fallback' suggestion ([#2419](https://github.com/thonny/thonny/issues/2149))