thonny/thonny #2148: Thonny not supported : Opening remote files in system app is not supported. Please download the file to a local directory and open it from there!

eya1903, 2022-01-31:
I am using thonny version 3.3.13, currently I am working with an ESP32 micropython. all my files are saved in the chip and also on my computer, they used to open and now this error appears : "Opening remote in system app is not supported.  Please download the file to a local directory and open it from there! " What is the issue please ? and how can I solve it ?
PS :  I can open the files saved on my computer but not the ones in the chip.

aivarannamaa, 2022-01-31:
Which extension did you use for the filenames when saving on the device?

eya1903, 2022-01-31:
I used python files ( .py ). I'm a newbie, this is my first experience in developpement,

aivarannamaa, 2022-01-31:
So you're double-clicking on a `.py`-file on your device and Thonny refuses to open it?

Which OS are you using?

aivarannamaa, 2022-01-31:
Please also check, what's the configuration for .py files in your Thonny -- right-click on a py file and select "Configure *.py files".

eya1903, 2022-01-31:
I'm sorry, My bad !! I configured to open th file with the 'system default app' instead of 'Thonny's text editor'. Now everything works good. Thank you for your time and effort ! :)

closed 2022-01-31 by aivarannamaa