thonny/thonny #2158: Provide a preference/option to "not program the on-chip RTC"

alanbchristie, 2022-02-06:
At the moment Thonny seems to always program the on-chip RTC (experienced with `3.3.13` on Raspberry Pi Pico)?

This appears to many to be helpful but, when working with external RTC modules or custom hardware (that can be used to setup the Pico RTC), having Thonny automatically re-program it is frustrating. Also, by automatically programming the RTC, code that uses the RTC may work when run using Thonny but not when not using Thonny. If users want the RTC wouldn't it be easier to let them program it themselves?

To satisfy everyone, can you add a preference/option (maybe under **Run and Debug**), to control it: -

- **Set on-chip RTC** (with a default value of Yes)

aivarannamaa, 2022-02-07:
Thank you for the suggestion!

Actually, this option already exist, it simply isn't presented in the configuration dialog. Please see https://github.com/thonny/thonny/wiki/MicroPython#advanced-configuration

In Thonny 4.0 this option will be presented in the configuration dialog as well (#1638)

alanbchristie, 2022-02-07:
Excellent! That works for me ... after I remembered NOT to edit the file while Thonny is running ;-)

Thanks - the existing configuration file feature is all I need so I'll close this issue and look forward to 4.0

closed 2022-02-07 by alanbchristie