thonny/thonny #2159: Fenster Tkinter croped

genpolf, 2022-02-06:
Hello, I have been using the Thonny IDE for some time. I'm surprised to find that my scripts that use Tkinter open a smaller sized graphics window than the one I specified. On the other hand, if I use the command "Run the current script in a terminal", the dimensions displayed are correct. There may be a configuration command that I missed or another way to solve this problem.
Thank you for your help.

aivarannamaa, 2022-02-07:
Thank you for the report!

I assume you are using Windows. Thonny marks the application process of user srcipts as DPI aware (`SetProcessDPIAware()`). Somehow it didn't occur to me that some users may not want it.

I'll add an option for this and make it off by default.

aivarannamaa, 2022-04-23:
I decided to skip automatic `SetProcessDpiAwareness` for user programs altogether. In order to get clear picture on modern displays on Windows, following lines should be added to the user program before creating Tkinter window:

```python
import ctypes
ctypes.OleDLL("shcore").SetProcessDpiAwareness(1)
```

closed 2022-04-23 by aivarannamaa