thonny/thonny #2162: autocomplete error

frozenhole, 2022-02-07:
[image]

aivarannamaa, 2022-03-23:
Thonny 4 has refactored autocompletion so you could try https://github.com/thonny/thonny/releases/tag/v4.0.0b1

abawer, 2022-10-07:
i followed to solution of @aivarannamaa  and removed then installed Thonny with the bash linux script. Autocomplete now works for me, but after auto-completion, i get the following error IDE error trace

`Traceback (most recent call last):
  File "/home/amit/apps/thonny/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/home/amit/apps/thonny/lib/python3.10/site-packages/thonny/plugins/autocomplete.py", line 171, in _on_select_item_via_event
    self._check_request_details()
  File "/home/amit/apps/thonny/lib/python3.10/site-packages/thonny/plugins/autocomplete.py", line 174, in _check_request_details
    assert self.winfo_ismapped()
AssertionError

`

aivarannamaa, 2022-10-07:
@abawer, thanks for the notice. I'll see if I can reproduce this.

abawer, 2022-10-07:
It only happened to me once, right after opening the IDE for the fresh install and trying to autocomplete (ctrl+space) over my code.

aivarannamaa, 2023-01-26:
I could not reproduce, but changed the code a bit that this error shouldn't happen again.

closed 2023-01-26 by aivarannamaa