thonny/thonny #2163: Thonny has no useable titlebar

nathanbern, 2022-02-09:
Using Thonny 3.3.14 on my pi 3 A+ with the latest OS, it does not have a titlebar, or the titlebar has disappeared behind the system toolbar. I have seen mention of fixes to this issue in pull requests but it appears it is on going. Just an FYI. See attached screenshot.

nathanbern, 2022-02-09:
[image]

nathanbern, 2022-02-09:
Screenshot is blurry in preview but looks OK if you open it.

aivarannamaa, 2022-02-11:
Thank you for the report!

Can you temporarily reduce your RPi taskbar size so that you get to move Thonny's window.

aivarannamaa, 2022-03-23:
Were you able to solve this issue?

nathanbern, 2022-03-23:
Hi! Sorry, I've been dealing with a bunch of other stuff and haven't been
on the software side of my job for a few weeks. If I remember correctly the
workaround I had to use was move the pi's taskbar/menu bar to the *bottom*
of the screen. Not the cleanest but it worked.

Thanks for following up!

NB

closed 2022-03-23 by aivarannamaa