thonny/thonny #2164: Package for Apple Silicon

andypiper, 2022-02-09:
At the moment, I'm able to run Thonny 3.3.13 on my M1 Pro MacBook Pro - it is the Intel build, running under Rosetta.

It would be great to have a native ARM build for Apple Silicon, or a universal binary for both Intel and Apple Silicon. The Python 3.10 build from Homebrew is compiled natively. I'm not sure what would need to happen for Thonny to run fully natively - I'm not sufficiently skilled with the Python packaging for macOS to know at the moment.

aivarannamaa, 2022-02-11:
Thank you for the report!

Thonny 4.0 will come with the universal build of Python 3.10.

aivarannamaa, 2022-03-23:
Here it is: https://github.com/thonny/thonny/releases/

closed 2022-03-23 by aivarannamaa