thonny/thonny #2167: Window resizing is wonky

bryanosborne, 2022-02-12:
When grabbing the upper left corner of the IDE to resize the pane, the lower right corner of the pane moves. The same aberration occurs when the lower left corner is grabbed and resizing attempted.

aivarannamaa, 2022-02-15:
Thank you for the report!

Whic OS are you using?

bryanosborne, 2022-02-15:
> CAUTION: THIS EMAIL ORIGINATED FROM OUTSIDE OF ORU

Aivar,

Thanks for the reply.

I found this on MacOS Monterey 12.2

Thanks,
Dr. O.

J. Bryan Osborne, D.C.S.
Assistant Professor of Computer Science
Director of ORU CITILab
Dept. of Computing and Mathematics
Oral Roberts University
7777 S. Lewis Ave.
GC 1 ½ E 06
Tulsa, OK 74171
Office Phone: (918) 495-6987
[CITILAB Seal Vector]

aivarannamaa, 2022-05-13:
Unfortunately I can't use Monterey on my Mac, but you could check whether the newer Tk in Thonny 4.0 helps: https://github.com/thonny/thonny/releases

AngryKiller, 2022-05-14:
> Unfortunately I can't use Monterey on my Mac, but you could check whether the newer Tk in Thonny 4.0 helps: https://github.com/thonny/thonny/releases

It does solve the issue for me on Monterey.

aivarannamaa, 2022-05-14:
Great! Thanks for letting me know!

bryanosborne, 2022-05-16:
This issue still exists in the 4.0.0b2 and is occuring on macOS Monterey running on an M1.

The issue should NOT be closed.

https://user-images.githubusercontent.com/22990921/168506828-8ddee590-5042-4c3b-88d0-7224bff8615c.mov

aivarannamaa, 2022-05-17:
@bryanosborne, thanks for the competing info! :)

I'll keep it open then, but I don't have ideas how to proceed with this. When my smaller child grows old enough to need a computer, I'll consider buying a M1 and try to reproduce this :D

AngryKiller, 2022-05-21:
I just upgraded to the new 14" MacBook Pro from my 13" Intel one, and I cannot reproduce the issue either on Thonny 4.0.0b2, that's weird. Also I was running Monterey 12.3.1 on the Intel and I'm now running 12.4 on the Apple Silicon.

https://user-images.githubusercontent.com/16365421/169645281-e9962dc4-a320-449a-b311-32d6d536a915.mov

bryanosborne, 2022-06-20:
I'm upgrading to 12.4.  Will check it after update.

bryanosborne, 2022-08-07:
Just going to add this to the mix...   Window resize problem occurs in 3.3.13 as well.  Please note this machine is an Intel chip, not an M1.  :-(
[image]

bryanosborne, 2022-08-07:
https://user-images.githubusercontent.com/22990921/183315017-c9a67023-45c1-488a-8c12-90a81bf55cdd.mov

aivarannamaa, 2024-05-29:
Fixed in 887ff2a8b4627bdf49f093ed3a99dfaf7fe10aef

closed 2024-05-29 by aivarannamaa