thonny/thonny #2171: Copy, paste broken.

Gorgus65, 2022-02-15:
Copy and paste does not work as it should, neither on Windows 10 or Raspberry Pi OS 64bit.

aivarannamaa, 2022-02-15:
Thank you for the report!

Unfortunately it is hard to fix this from this description. Can you give me the steps for reproducing the problem? What did you do, what did you expect to happen and what did happen?

Gorgus65, 2022-02-15:
If I have a small program, mark a few rows and press ctrl-c or use the menu to copy, Thonny instead do a ctrl-v or paste whatever is in the buffer. The marked text will not end up in the buffer and it disappears from the editor. It is like it does a ctrl-x (without sving the text) and immediately after that does a paste.

Ctrl-x or cut works as it should and I can paste that correctly.

Otherwise I am very happy with using  Thonny, for me it is the best.

Regards
Göran Gustafsson

aivarannamaa, 2022-02-16:
That's really weird... Which Thonny version are you using?

Gorgus65, 2022-02-16:
3.3.13 on Win10 and 3.3.14 on Raspberry Pi OS (Debrani GNU/Linux 11).

/GG

aivarannamaa, 2022-02-21:
Did you happen to use the middle mouse button while experiencing this problem?

Gorgus65, 2022-02-21:
No, just the right one.

aivarannamaa, 2022-03-23:
Please try, whether https://github.com/thonny/thonny/releases/tag/v4.0.0b1 works better

aivarannamaa, 2022-03-23:
Possibly related to #1934

Gorgus65, 2022-10-11:
It is working great now using the keyboard but still the same when using rightclick or menus. Sorry.

aivarannamaa, 2022-12-28:
Unfortunately I can't reproduce this.

closed 2022-12-28 by aivarannamaa