thonny/thonny #2177: raspbery pi pico: cant program a py to flash

edprest, 2022-02-18:
Newby user. Demo program runs Ok in Thonny.
 When trying to **save a main.py file to pico run from flash,** can't save. Windows pop up for filename and this happens...
info:
    Running on ubuntu 20.04.3 
    Thonny 3.2.7
    python 3.8.10
    tk 8.6.10

rp2-pico-20220117-v1.18.uf2       on raspberry pi pico board

I get ...

Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 969, in dispatch_from_menu
    dispatch(None)
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 911, in dispatch
    handler()
  File "/usr/lib/python3/dist-packages/thonny/code.py", line 745, in _cmd_save_file_as
    self.get_current_editor().save_file(ask_filename=True)
  File "/usr/lib/python3/dist-packages/thonny/code.py", line 221, in save_file
    save_filename = self.ask_new_path()
  File "/usr/lib/python3/dist-packages/thonny/code.py", line 309, in ask_new_path
    return self.ask_new_remote_path()
  File "/usr/lib/python3/dist-packages/thonny/code.py", line 312, in ask_new_remote_path
    target_path = ask_backend_path(self.winfo_toplevel(), "save")
  File "/usr/lib/python3/dist-packages/thonny/base_file_browser.py", line 1157, in ask_backend_path
    dlg = BackendFileDialog(master, dialog_kind, proxy.get_cwd())
  File "/usr/lib/python3/dist-packages/thonny/base_file_browser.py", line 1016, in __init__
    self.browser.request_focus_into(initial_dir)
  File "/usr/lib/python3/dist-packages/thonny/base_file_browser.py", line 933, in request_focus_into
    assert path == ""
AssertionError

------------------------
drag and drop main.py does not write either.

edprest, 2022-02-19:
**work around**  is to select generic python interpreter , then saving to flash works.

aivarannamaa, 2022-02-20:
Thank you for the report!

Can you reliably reproduce this problem with "MicroPython (Raspberry Pi Pico)" back-end?

edprest, 2022-02-21:
100 percent of the time. The cause is that I did not see a PICO choice in
the  Run - Select Interpreter pull down so i used the first good looking
choice (BBC) which runs but wont store.
When I run as generic it stores fine.

So, the issue is really that:   PICO does not show up in pull down
interpreter list.
Screen shot attached with all choices I have.  Also see the shell showing
output from flash stored python.

aivarannamaa, 2022-02-21:
Oh, sorry.. I only now noticed that you had Thonny 3.2.7. It's rather old version.

I know that Ubuntu's repo has outdated version and I don't consider it my responsibility. My suggeston is to apt-remove thonny and install latest Thonny+Python bundle instead:

```
$ bash <(wget -O - https://thonny.org/installer-for-linux)
```

edprest, 2022-02-21:
ran uninstall and then installed thonny 3.3.13   pico appears as a
selection and runs ok. Still cant program to flash other than the first
time  yesterday    main.py

I have run the flash_nuke.uf2  and reloaded the rp2-pico-20220117-v1.18.uf2
file - Ok.
main.py runs in thonny fine and every time.
cant program to pico flash by copy or drag and drop to rpi-rp2 dir

edprest, 2022-02-24:
Closing as resolved. Newby got confused with old Thonny and then with PIC devices which do program by dropping into dir. I said I was a newby.

aivarannamaa, 2022-02-28:
I'm glad you found a solution!

closed 2022-02-28 by aivarannamaa