thonny/thonny #2178: Pi pico output not displaying in shell but its working with "putty", and i also cant read from pi pico

OpGamer3000, 2022-02-19:
So i was writing my code as usual, after adding a while loop (which has no break), i can no longer upload the code or read my previous code from it
when i try it to do then i get this error:

Device is busy -- can't perform this action now.
Please wait or cancel the current work and try again!
[image]

there are no outputs, no shell stuff, etc
idk what to do now

any help will be greatly appreciated
and also suggest some precautions to take if possible...

OpGamer3000, 2022-02-19:
so i managed to get the output from pi pico but stil cant access the dile system in pi pico

OpGamer3000, 2022-02-19:
so i decided to nuke(reset) the flash instead, but this is a problem(or maybe i am doing something wrong). please fix this or suggest precautions.

OpGamer3000, 2022-02-20:
so we can use rshell(linux or WSL) to do recover our files(worked for me!), stil if you guys have other sugestions then tell me

aivarannamaa, 2022-02-20:
Thank you for the report!

Did you try interrupting your loop with Ctrl+C ?

OpGamer3000, 2022-02-22:
> Thank you for the report!
> 
> Did you try interrupting your loop with Ctrl+C ?

yes i did but dint work

OpGamer3000, 2022-02-24:
i'll just say that avoid using all the cores, if u have to do something forever then use the second core.

closed 2022-02-24 by OpGamer3000