thonny/thonny #2180: Thonny Broken - Can't Fix It -  HELP!!!

cpwoodrum, 2022-02-22:
System:  Raspberry Pi 4 running Rasberian. 

Trying to run:

print("hello")

I get: 

WARNING:root:Unexpected echo. Expected b'%Run -c $EDITOR_CONTENT\r\n', got b''

Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - use Stop/Restart to interrupt more and enter REPL.
  
I have tried: 
removing and reinstalling using the Add/Remove Software feature.
installed an earlier backup version where I know thonny was working.

Neither fixed the problem.

Python IDLE works ok.

Can anyone help?

pasdechance, 2022-02-23:
There are similar reports with the same error. 

Have any of the solutions or recommendations in these issues helped?

https://github.com/thonny/thonny/issues/1668 / https://github.com/thonny/thonny/issues/1743

cpwoodrum, 2022-02-23:
Hello Jacques -

After a great deal of trial, error, and colorful language, here's what I
found.  I had been programming an Arduino Pico with CircuitPython.  To do
that with Thonny, I had to go to Tools/Options/Interpreter and change the
interpreter to CircuitPython.  When I changed to an Arduino Nano, I failed
to change the interpreter back to the default (The same interpreter that
runs Thonny).  After changing the interpreter back to that, the problem
went away.

Perhaps this should be posted to help others?

Kind regards.

aivarannamaa, 2022-02-28:
Thank you for the report and sorry for the late reply. This kind of error happens when you try to run a script but Thonny is not in proper state for that. It will be fixed in the upcoming version 4.0

closed 2022-02-28 by aivarannamaa