thonny/thonny #2181: BLD: Install thonny with conda and/or mamba

westurner, 2022-02-22:
User story: Users can install Thonny with conda and/or mamba:

```bash
conda install -c conda-forge -y thonny

conda install -c conda-forge -y mamba

mamba install -c conda-forge -y thonny
mamba install -y thonny
```

- [ ] Create a conda-forge feedstock for thonny
  - [ ] Create a separate conda-forge feedstock(s) for thonny plugin(s)

Docs:
- https://conda-forge.org/docs/
  - https://conda-forge.org/docs/maintainer/adding_pkgs.html
  - https://conda-forge.org/docs/maintainer/updating_pkgs.html
    > Pushing to regro-cf-autotick-bot branch¶
    > When a new version of a package is released on PyPI/CRAN/.., we have a bot that automatically creates version updates for the feedstock. In most cases you can simply merge this PR and it should include all changes. When certain things have changed upstream, e.g. the dependencies, you will still have to do changes to the created PR. As feedstock maintainer, you don’t have to create a new PR for that but can simply push to the branch the bot created. There are two alternatives […]
- https://github.com/conda-forge/spyder-feedstock/tree/master/recipe

westurner, 2022-02-22:
(also FWIU, the Thonny Win/Mac/Lin installers could instead be built with https://github.com/conda/constructor [in e.g. GitHub Actions]. IDK what the pros/cons of that separately scoped issue would be?)

aivarannamaa, 2022-02-28:
Sorry for the late reply!

Do I understand correctly, that this is a feature request for making Thonny installable with conda?
Isn't it already possible to pip-install Thonny into a conda environment?

westurner, 2022-02-28:
Np. Not a high priority for me at present.

> By default, we build packages for macOS, Linux AMD64 and Windows AMD64.

- [ ] DOC: Also:
  - *linux-arm64* (ARM64; [Raspberry Pi, Rockchip,] SBCs) it should say:
  - *osx-arm64*
    - https://conda-forge.org/blog/posts/2020-10-29-macos-arm64/#how-to-add-a-osx-arm64-build-to-a-feedstock

https://conda-forge.org/docs/maintainer/infrastructure.html#compilers-and-runtimes

https://conda-forge.org/docs/maintainer/infrastructure.html#conda-forge-admin-please-rerender

https://conda-forge.org/docs/maintainer/adding_pkgs.html#the-staging-process

https://github.com/conda-forge/conda-smithy-feedstock/blob/master/recipe/meta.yaml 

https://github.com/conda-forge/conda-smithy-feedstock/blob/master/README.md :

***

Terminology
===========

**feedstock** - the conda recipe (raw material), supporting scripts and CI configuration.

**conda-smithy** - the tool which helps orchestrate the feedstock.
                   Its primary use is in the construction of the CI ``.yml`` files
                   and simplify the management of *many* feedstocks.

**conda-forge** - the place where the feedstock and smithy live and work to
                  produce the finished article (built conda distributions)

closed 2022-03-23 by aivarannamaa

villares, 2022-03-28:
I see how this could be useful... 

The `py5` library I'd like to use in my classrooms depends on `jpype1` that lately seems to fail to install with `pip` on Windows, but it seems to install alright with `conda`.

So if I could instruct the students at home or the lab maintainers to download `miniconda` and with  **Thonny** on *conda-forge*, it might reduce the number of steps they have to execute. Maybe I could even make some hackish script like this: 

```python
import conda.cli
conda.cli.main('conda', 'install',  '-y', 'thonny)
conda.cli.main('conda', 'install',  '-y', 'py5')
```

westurner, 2022-03-28:
> my other option will be to try to create a portable version with everything pre-installed 😄

https://github.com/conda/constructor would be easy to do, but work to maintain

If someone creates a feedstock with these instructions and the appropriate contact info, I'd be happy to look it over and send PRs if even needed. Conda forge auto-rebuilds when there's a new version pushed to pypi.
https://conda-forge.org/docs/maintainer/adding_pkgs.html#the-staging-process (and other links above)

aivarannamaa, 2022-03-28:
I tried to install thonny-py5mode with latest Thonny beta (containing 64-bit Python 3.10) and could confirm the problem with jpype1. IMO the best solution in this case would be convincing jpype1 developer to publish 64-bit wheels for Python 3.10.

westurner, 2022-03-28:
>  publish 64-bit wheels for Python 3.10.

https://github.com/pypa/cibuildwheel

https://anaconda.org/conda-forge/jpype1

https://anaconda.org/conda-forge/thonny

https://github.com/conda-forge/spyder-feedstock/tree/master/recipe