thonny/thonny #2183: Cannot open system shell

pasdechance, 2022-02-23:
Hello,

When I try to launch Tools > Open System Shell, it opens and informs me that:

```
Failed to launch child: bash -c "/usr/bin/python3 /usr/lib/python3/dist-packages/thonny/plugins/system_shell/explain_environment.py; exec bash -i"
With error: No such file or directory
Press Enter to exit.
```

Thonny 3.3.14 on Debian, installed via apt. I also tried installing via pip and received the same error. I suspect the problem is on my end.

Any thoughts?

Thanks in advance.

pasdechance, 2022-02-24:
**Additional Information**

DE: LXQt / Terminal: kitty

Expected behaviour:

Output: 
```
********************************************************************************
Some Python commands in the PATH of this session:
 - python3   -> /usr/bin/python3.9
 - pip       == /home/pasdechance/.local/bin/pip
 - pip3      == /home/pasdechance/.local/bin/pip3

********************************************************************************
```

Workaround:

run `update-alternatives --config x-terminal-emulator` as per https://github.com/thonny/thonny/issues/1129#issue-587180103

Output:
```
There are 6 choices for the alternative x-terminal-emulator (providing /usr/bin/x-terminal-emulator).

  Selection    Path                                 Priority   Status
--------------------------------------------------------------------------
  0            /usr/bin/qterminal                   40        auto mode
* 1            /home/pasdechance/.local/bin/kitty   0         manual mode
  2            /usr/bin/koi8rxterm                  20        manual mode
  3            /usr/bin/lxterm                      30        manual mode
  4            /usr/bin/qterminal                   40        manual mode
  5            /usr/bin/uxterm                      20        manual mode
  6            /usr/bin/xterm                       20        manual mode
```

- [ ] **qterminal**: flashes, but does not remain open
- [ ] **kitty**: opens with error
- [ ] **koi8rxterm**: didn't bother trying
- [X] **uxterm**: works
- [X] **lxterm**: launches uxterm but displays information correctly
- [X] **xterm**: launches uxterm, displays information

**Conclusion: I need to change something to make it work with kitty, but I don't know where to look to do that**

aivarannamaa, 2022-02-28:
Thank you for the report!

I'm afraid I don't find time for supporting less used terminal emulators as each of them may need different approach for submitting commands and for staying open. 

I encourage you to try to add support for your terminal. The relevant code is here: https://github.com/thonny/thonny/blob/3f34597a6c98c120575e65a69dc96233371f07e1/thonny/terminal.py#L70

pasdechance, 2022-03-02:
I'll close this issue. I have found that other programs react the same way, not just Thonny. If I find something that works, I will post a solution or make a PR.

closed 2022-03-02 by pasdechance