thonny/thonny #2184: Add config to allow pip --trusted-host pypi.org  --trusted-host files.pythonhosted.org

jobartucz, 2022-02-25:
Would it be possible to allow a config where pip uses --trusted-host pypi.org  --trusted-host files.pythonhosted.org

In school environments, pip can fail with 

self-signed certificate in chain.

aivarannamaa, 2022-02-26:
Nevermind, normal pip usage is not cocerned with micropython.org, whose certificate chain is the problem in #1986

Givy, 2022-02-26:
Please see below Attached Error:
[image]

Thank you.

Givy, 2022-02-26:
[image]

Givy, 2022-02-26:
Hi,
I hope these help:

[image]

Thank you.

aivarannamaa, 2022-02-27:
@Givy, thonny-rpi.pico plugin is obsolete. In the future please create a separate issue for separate problems. This issue is not suitable for for discussing your problem.

jobartucz, 2022-02-27:
> Have you tried specifying these options via environment variables (Tools => Options => General)? See https://pip.pypa.io/en/stable/topics/configuration/#environment-variables

but since the arguments are the same, how could I have two environment variables with the same name point to two different places?

PIP_TRUSTED_HOST="files.pythonhosted.org"
PIP_TRUSTED_HOST="pypi.org"

Is there a way to combine them?

Givy, 2022-02-27:
Hi,
OK..  "thonny-rpi.pico plugin is obsolete.", thank you...But, what needs to replace it?  I have not modified anything in Thonny since I installed it!!  
I have opened a couple of new issues.  Please also see "Thonny 3.3.13 can't interpret #2187".
Thank you.

aivarannamaa, 2022-02-28:
> Is there a way to combine them?

Did you try separating two values with space, as with PIP_FIND_LINKS example at https://pip.pypa.io/en/stable/topics/configuration/#environment-variables ?

jobartucz, 2022-02-28:
It didn't work. :(

[image]

KennyHarrer, 2022-02-28:
First and foremost its worth mentioning Dashes (``-``) have to be replaced with underscores (``_``) ([see this](https://pip.pypa.io/en/stable/topics/configuration/#environment-variables)). Secondly the secondly ``--trusted-host`` overwrites the first. I am on the same network as @jobartucz , adding ```PIP_TRUSTED_HOST="files.pythonhosted.org"``` did not fix the issue.

aivarannamaa, 2022-02-28:
Please try without quotes. Thonny simply splits the line on `=` and sets the rightmost part as value of the variable

KennyHarrer, 2022-02-28:
That solved the issue! Thank you. I will relay this information to @jobartucz .

closed 2022-03-23 by aivarannamaa