thonny/thonny #2185: INSTALL MicroPython Firmware on RPI-PICO RP2 mounted on h:\  ;;SSL: CERTIFICATE_VERIFY_FAILED

Givy, 2022-02-25:
Hi,
Getting below issues:

Downloading 563200 bytes from https://micropython.org/resources/firmware/rp2-pico-20210902-v1.17.uf2

Traceback (most recent call last):
  File "S:\Thonny Python IDE\lib\urllib\request.py", line 1350, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "S:\Thonny Python IDE\lib\http\client.py", line 1277, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "S:\Thonny Python IDE\lib\http\client.py", line 1323, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "S:\Thonny Python IDE\lib\http\client.py", line 1272, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "S:\Thonny Python IDE\lib\http\client.py", line 1032, in _send_output
    self.send(msg)
  File "S:\Thonny Python IDE\lib\http\client.py", line 972, in send
    self.connect()
  File "S:\Thonny Python IDE\lib\http\client.py", line 1447, in connect
    server_hostname=server_hostname)
  File "S:\Thonny Python IDE\lib\ssl.py", line 423, in wrap_socket
    session=session
  File "S:\Thonny Python IDE\lib\ssl.py", line 870, in _create
    self.do_handshake()
  File "S:\Thonny Python IDE\lib\ssl.py", line 1139, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1091)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "S:\Thonny Python IDE\lib\site-packages\thonny\plugins\micropython\uf2dialog.py", line 271, in _perform_work
    self._download_to_the_device(download_url, size, target_dir)
  File "S:\Thonny Python IDE\lib\site-packages\thonny\plugins\micropython\uf2dialog.py", line 333, in _download_to_the_device
    with urlopen(req, timeout=5) as fsrc:
  File "S:\Thonny Python IDE\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "S:\Thonny Python IDE\lib\urllib\request.py", line 525, in open
    response = self._open(req, data)
  File "S:\Thonny Python IDE\lib\urllib\request.py", line 543, in _open
    '_open', req)
  File "S:\Thonny Python IDE\lib\urllib\request.py", line 503, in _call_chain
    result = func(*args)
  File "S:\Thonny Python IDE\lib\urllib\request.py", line 1393, in https_open
    context=self._context, check_hostname=self._check_hostname)
  File "S:\Thonny Python IDE\lib\urllib\request.py", line 1352, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1091)>

Givy, 2022-02-25:
UF2 Bootloader v1.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2

aivarannamaa, 2022-02-26:
Please see https://github.com/thonny/thonny/issues/1986

closed 2022-02-26 by aivarannamaa

Givy, 2022-02-26:
Please see below Attached Error:
[image]