thonny/thonny #2187: Thonny 3.3.13 can't interpret

Givy, 2022-02-27:
Hi,
I am not sure what the issue is, but I have tried to update thonny plugins as much as possible.  However, simple commands are having issues.  Please see below.  What am I missing?

Adafruit CircuitPython 7.2.0 on 2022-02-24; Raspberry Pi Pico with rp2040
>>> from machine import Pin
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'machine'
>>> led = Pin(25, Pin.OUT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'Pin' is not defined
>>> 

Thank you.

aivarannamaa, 2022-02-28:
CircuitPython does not have `machine` module. It is present only in MicroPython.

closed 2022-02-28 by aivarannamaa

Givy, 2022-02-28:
Can you tell me where these "modules" are and how I can update them from a list if there is one?  Thanks.

Givy, 2022-02-28:
Adafruit CircuitPython 7.2.0 on 2022-02-24; Adafruit Grand Central M4 Express with samd51p20
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 8, in <module>
ImportError: no module named 'adafruit_character_lcd'
>>>

aivarannamaa, 2022-02-28:
These questions are better asked at CircuitPython forum or MicroPython forum