thonny/thonny #2188: Thonny showing Backend terminated

sahabitukur01, 2022-03-01:
My Thonny keeps on displaying "
Backend terminated or disconnected. Use 'Stop/Restart' to restart."
even after Stop/Restart,please Help !!

aivarannamaa, 2022-03-01:
Which Thonny version are you using? Which OS? Which back-end (Python 3 or MicroPython)? Did it start displaying this error right after you launched Thonny for the first time or after you did something in Thonny?

sahabitukur01, 2022-03-02:
I am using Thonny current version 3.3.13 on a windows 10 operating system and back-end is Python 3.It started showing errors  after using it for a while.

aivarannamaa, 2022-03-02:
What do you have written on the lower-right corner of your Thonny's main window?

aivarannamaa, 2022-03-02:
even better -- could you post the screenshot of your Thonny's main window, together with the error message?

aivarannamaa, 2022-03-23:
Please reopen if you have more information to add

closed 2022-03-23 by aivarannamaa