thonny/thonny #2189: ModuleNotFoundError: No module named 'serial.tools'

Jawad-Alhamadah, 2022-03-03:
the issue started with thonny giving me an error  in the shell about not finding the serial module. After looking around, i did the only two suggested solutions: change any file that is named serial(i didn't have any), pip install serial.  After installing serial, thonny started giving me an error  in a window titled " error initializing backend": 
`ModuleNotFoundError: No module named 'serial'`

i tried installing serial using pip and that's when the issue changed to:
`ModuleNotFoundError: No module named 'serial.tools'`

 I have searched online for solutions and it seems that all the suggestions don't work. I ran all kinds of pip installs of serial and pyserial.   Im unsure what to do to resolve this issue. I hope you guys can help. Thank you.

im using Raspberry pi 4.

aivarannamaa, 2022-03-05:
`pip install serial` messes up things. Thonny needs distribution package `pyserial` (which contains Python module-packages `serial` and `serial.tools`.

So try uninstalling all other serial-like packages and keep only pyserial.

aivarannamaa, 2022-03-05:
> the issue started with thonny giving me an error in the shell about not finding the serial module.

Did you try to `import serial` in your MicroPython code? AFAIK, there is no such module which would work with MicroPython. Pyserial is meant for CPython.

aivarannamaa, 2022-03-23:
Please reopen if you have more details to add

closed 2022-03-23 by aivarannamaa