thonny/thonny #2190: issue Thonny vers. 3.3.13 (Windows 10 (64-bit)) (Raspberry Pi PICO)

jsbrodhead, 2022-03-03:
Hello,
As my first program (script) in Thonny has grown, I am experiencing an error pop-up, with increasing and predictable occurrence:

 Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 1712, in event_generate
    handler(event)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 1457, in _on_output
    self.append_text(msg["data"], stream_name)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workdlg.py", line 243, in append_text
    setattr(self, stream_name, getattr(self, stream_name) + text)
MemoryError

This appears upon the first message "print("Reading Temp & RH")" and APPEARS to be occurring on every iteration of print( ).  To get past the error message, I must disconnect the PICO (USB) and click the close button repeatedly, until all messaged in the queue have made their way to the shell window.

My entire program is about 600 lines (including white-space line and comment-only lines.  I have uS pulse measurement, MODBUS RTU Master (temperature and relative humidity monitoring), (look-up list) conversion to dew-point, digital inputs and outputs all working working.  Just a couple more devices (I2C EEPROM and a MODBUS RTU Slave port) to go and the project is completed... (with zero free pins on the PICO).  Diagnostics is SOOOOO much easier with print( ) as my eyes to the steps along the way.

Can anyone answer if there is a revision I should revert to, to avoid this issue, or something different in configuration?  I thought of creating modules, or a packet of modules, in order to shed bulk in main.py, but this is my first (micro)python; first Raspberry Pi project ever, and adding one more layer of complexity for me, looks like more potential for programming mistakes - though I will consider that course, if necessary.

Thank you,
Regards,
Jeff Brodhead

jsbrodhead, 2022-03-03:
As I wrote in my initial comment (issue report), regarding MOVING MY FUNCTIONS TO A SEPARATE FILE - that worked with the very first function removed from main.py into funcs.py.  Other than duplicating a few variable defines, module imports, changing a couple globals into passed parameters (better anyway), no more pop-up error reports / issues with print( ) calls sticking in queue.

aivarannamaa, 2022-03-05:
Thank you for the report!

Can you share your code?

jsbrodhead, 2022-03-07:
The issue continued, as my program grew even just a little, so my son said I should try Visual Studio Code and PICO-Go, which I did.  He uses Visual Studio Code at the university.  I am ready for a dog&pony (demonstration) today.  Micropython got me quite a way into the project, but just could not handle everything I threw at it.

Visual Studio Code and PICO-Go turned out to be a great alternative - my Micropython code dropped in and ran as-was.  I don't have time to turn my issues with Micropython into a debug project for Micropython.

To anyone reading my comments, Micropython got me a long way into my project and I'm sure that for most teaching environments, where people are learning to program, it is a great tool!  It has a memory limitation I couldn't diagnose, fix, or get past, except to transition to another .

This is my first RP2040 project - I was going to use a PIC16F1829 (programmed in the old Microchip C (IDE 8.92), which I have been using for many years).  I have 100 of the PICs in-hand and even have a PICO pin-compatible'ish PCB designed for it, but the PICO has proven to be quite useful - providing more functionality, with a one-week turnaround, from zero to nearly completed program (script).

As I am writing this code for a customer, I don't think I can justify sharing the entire program.  A lot of it wouldn't make sense without the schematic and the logic behind it (A friend taught me many years ago "Improvise, Adapt and Overcome" (IA&O) and that is the way MOST of my designs go...  

>>>Are you interested in any particular portion(s) of my code?<<<

aivarannamaa, 2022-04-10:
Please try https://github.com/thonny/thonny/releases/tag/v4.0.0b2 which has several fixes related to MicroPython, and also https://micropython.org/resources/firmware/rp2-pico-20220410-unstable-v1.18-307-g66b5c4c77.uf2 which has a nasty bug fixed in https://github.com/micropython/micropython/commit/587339022689187a1acbccc1d0e2425a67385ff7

aivarannamaa, 2023-01-28:
Please try the newest Thonny version (4.0.2 as of now) and re-open if you still have this error.

closed 2023-01-28 by aivarannamaa