thonny/thonny #2192: Error with multiprocessing with Thonny 3.3.13 under Windows10

yvesg54, 2022-03-06:
Running this code fails.
```
import multiprocessing 
def y(x):
    print(x)
if __name__ == '__main__':
    f = multiprocessing.Process(target = y, args=('a'))
    f.start()
    f.join()
```Errors:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Program Files (x86)\Thonny\lib\multiprocessing\spawn.py", line 105, in spawn_main
    exitcode = _main(fd)
  File "C:\Program Files (x86)\Thonny\lib\multiprocessing\spawn.py", line 115, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'y' on <module '__main__' (built-in)>
```

aivarannamaa, 2022-03-06:
Thank you for the report!

In order to support several features (including the "nicer" debugger), Thonny wraps your code rather heavily and this doesn't play nice with the new processes created by `multiprocessing`.

I'm afraid I can't do anything about this.

The work-around is to use "Run => Run current script in terminal" for such programs, which runs the script in traditional way.

yvesg54, 2022-03-06:
Thank you for your feedback, and for the perfect workaround.
I installed PyCharm to test my program, but, indeed, running it in Thonny terminal mode is a perfect alternative, allowing me to use the great capabilities offered by Thonny.
Would it be possible to get a message stating that multiprocessing requires using the terminal mode of Thonny?  Or perhaps documenting that limitation somewhere?  I made a search "thonny multiprocessing", and did not obtain any clear statement.
Thank you very much for allowing us to benefit from a nice and efficient tool such as Thonny!

aivarannamaa, 2022-03-23:
I'll add a warning and suggestion for running via Terminal when Thonny's backend detects importing multiprocessing.

aivarannamaa, 2022-03-24:
I'll test later, whether a6bafccf6a584bfdf6b2de81795ed9758b44615e helped here as well

aivarannamaa, 2022-04-23:
It looks `multiprocessing` works in Thonny 4.0.0b2

closed 2022-04-23 by aivarannamaa

Ivoz, 2022-08-16:
> I'll test later, whether [a6bafcc](https://github.com/thonny/thonny/commit/a6bafccf6a584bfdf6b2de81795ed9758b44615e) helped here as well

Applying this just in my 3.3.13 thonny worked as well, to get multiprocessing working. Line 133 under `Thonny\Lib\site-packages\thonny\plugins\cpython\cpython_backend.py`