thonny/thonny #2193: file or folder not writable problem

plotzz, 2022-03-07:
i searched everywhere on youtube and other websites to no avail pls help

aivarannamaa, 2022-03-07:
I don't know this problem. Can you elaborate?

plotzz, 2022-03-08:
> I don't know this problem. Can you elaborate?

it goes like this when i use thonny i open it and try to use python i then try to save it and it a notification appears saying "this file or folder is not writable
[image]
"

aivarannamaa, 2022-03-08:
Where are you trying to write the file? Under your home directory or somewhere else?

aivarannamaa, 2022-03-23:
Please reopen if you have more details to add!

closed 2022-03-23 by aivarannamaa

JerichoTTV, 2022-12-21:
I am having this same problem, I'm currently trying to open the Clangen patrol rework through thonny but each time it gives me that same error message, I have all the requirements; Pygame, Pygame-gui, and ujson

Youarenotblind, 2023-05-23:
> i searched everywhere on youtube and other websites to no avail pls help

去掉这个文件的“只读”属性就可以了

Blank2275, 2023-07-12:
This happened to me, the solution was I had to change 'open(file, "r")' to 'open(file, "w")'