thonny/thonny #2194: Webrepl not available on some installs for the same version

prgrobots, 2022-03-08:
Hello

I have the same version of Thonny installed on two different windows 10 PCs, one has the webrepl available where the port drop down box is and one doesn't?
[image]
I tried re-installing but its still not there ,and kinda awesome to have a webrepl inside the IDE.

prgrobots, 2022-03-09:
I have installed the latest, 3.3.1.3 but its only installed 3.2.1

[image]

My other machine is running 3.3.1.3 and has the webrepl

aivarannamaa, 2022-03-09:
In this case you have an older instance of Thonny hiding the newer one.

Please go to "Tools => Manage plug-ins" and look for `thonny`. If it shows version 3.2.1, then uninstall it and restart Thonny.

aivarannamaa, 2022-03-23:
Please reopen if you still have this problem

closed 2022-03-23 by aivarannamaa