thonny/thonny #2196: Thonny 3.3.13 crashes on ChromeOS Crostini (fresh install)

coloneldare, 2022-03-12:
Until yesterday Thonny 3.3.13 was running RaspberryPi micropython beautifully, stable and useful in a Crostini Linux (Buster) container on my Chromebook (OS #99) prior to latest updates (to #100).
(Previously in early February)
[image]

(Now using ASUS C433T)
Powerwash => reinstall (Bullseye) then the same 'bash <(wget -O - https://thonny.org/installer-for-linux)'
.....
```
Installation was successful, you can start Thonny from start menu under
Education or Programming, or by calling /home/******/apps/thonny/bin/thonny
For uninstalling Thonny call /home/******/apps/thonny/bin/uninstall
```

Unfortunately calling Thonny, directly, or from the menu seems to do nothing, or places a zero size window on the desktop (looks like a vertical cursor of zero width.  Pulling it open shows the usual Thonny window but this collapses as soon as any menu option is selected:  e.g:

```
/home/******/apps/thonny/bin/thonny
INFO    thonny: Thonny version: 3.3.13
X connection to :0 broken (explicit kill or server shutdown)
```

Note: Other packages seem to run fine.e.g. LibreOffice, Gimp, Terminal etc.

Great product thank you, but sad it's broken for me at the moment.

aivarannamaa, 2022-03-16:
Thank you for the report!

I updated my Chromebook to latest version (had updated Crostini to Bullseye already earlier), but could not reproduce your issue. 

Can you run Thonny when you `apt install thonny`?

I'll soon publish the first beta of Thonny 4.0, with binary bundle containing newer Tcl/Tk -- hopefully this will work for you.

coloneldare, 2022-03-16:
Thank you Aivar for taking my report on board, much appreciated.

I did another Powerwash and turned on 'developers/linux'.  When finished
installing I did apt update (6 items) and immediately apt install thonny
(i.e before I put any other clutter in there like Libreoffice, gimp etc).

-> 'permission denied' (unsurprisingly) so sudo apt install thonny seemed
to work.  But it didn't (same problem!)

To preserve the terminal install log I pasted it into a Google document (29
pages) and towards the end it said:

update-alternatives: using /usr/share/docutils/scripts/python3/rst2xml to
> provide /usr/bin/rst2xml (rst2xml) in auto mode
> update-alternatives: using /usr/share/docutils/scripts/python3/rstpep2html
> to provide /usr/bin/rstpep2html (rstpep2html) in auto mode
> Setting up thonny (3.2.7-1) ...
> Processing triggers for dictionaries-common (1.28.4) ...
> aspell-autobuildhash: processing: en [en-common].
>
[cut]

> aspell-autobuildhash: processing: en [en_US-wo_accents-only].
> ***@***.***:~$
>

From previously helping a friend we found 3.2.7 didn't support the Raspi
Pico but 3.3.13 (using wget from the Thonny homepage) did.

Most curious - I am out of my depth but I find both routes (apt install
from Debian, or the wget installer) both fall over as soon as I try to open
a menu (such as 'options' or 'about') - it just vanishes in a puff of
virtual smoke!

If I can help further by retrieving any error logs I will happily, but I
will need some guidance as to what and how to find them.

Kind regards, and thanks for a brilliant product.

btw - I still have access to Thonny via my RasPi 400 so not all is lost.

aivarannamaa, 2022-03-16:
I'll let you know when the first beta of 4.0 is out so you can check whether it makes a difference.

In the meanwhile you could try whether Python's IDLE works for you. It uses the same UI toolkit as Thonny.

coloneldare, 2022-03-16:
>
> ....you could try whether Python's IDLE works for you.

Nope.  :-(
Just loaded it and it crashes exactly the same (zero width window, like a
vertical cursor which when dragged open has some functionality):

Python 3.9.2 (default, Feb 28 2021, 17:03:44)
> [GCC 10.2.1 20210110] on linux
> Type "help", "copyright", "credits" or "license()" for more information.
> >>> 22/7
> 3.142857142857143
> >>>

 ... but self destructs as soon as  a menu option is selected.  Menus seem
to drop down OK, but selecting an option crashes it.  Curiously 'drag,
ctrl+c' copied the text above just fine, but selecting menu 'edit' > 'copy'
=> 'puff, gone'.

nb, I'll be happy to play with 4.0 beta when available if that would be
helpful.

Thanks,  Doug.

coloneldare, 2022-03-16:
> Just loaded it and it crashes exactly the same
>
> In this case it's worth reporting the issue at https://bugs.python.org --
> there are lots of smart people who may have good ideas. I'm pretty sure
> they want IDLE to work properly on Chromebooks.

Thanks

coloneldare, 2022-03-22:
> You could now try the first beta of 4.0:

Cool, thanks, I'll give it a go a bit later.
Best, Doug.

coloneldare, 2022-03-24:
Sadly I am still stuck with the same issue.  I'm not technical enough to
find out where the process goes when it collapses after picking a menu
item.  It seems to me the UI falls over but Thonny is still running (I
suspect) - having had it go 'puff' and vanishing it won't re-run from the
icon, nor does the 'zero size window' appear unless I close the linux
container with shutdown then restart Thonny from the main menu.

I'll happily chase any clues if you tell me how, but for now I have
recorded a screen movie so you can hopefully see what I see.

All the best, Doug
ps I like the Ukraine flag - a nice touch!  :-)

aivarannamaa, 2022-03-26:
I'm going to close this as I can't reproduce the problem and don't have any ideas how to go further. If you want to solve this, then I recommend lookin for the solution to IDLE first, as it has larger and more knowledgeable team. Most likely I can take over their solution then.

closed 2022-03-26 by aivarannamaa

aivarannamaa, 2022-04-09:
I could reproduce the problem after re-installing Crostini. At bugs.python.org you mentioned having reported the problem at Google. Can you share the link?

coloneldare, 2022-04-09:
Good to hear it's not just me having problems.
The bug reporting method was as per:
https://support.google.com/chromebook/answer/2982029?hl=en-GB
I.e holding ALT+SHIFT +'i''. Which brought up a dialogue box to fill stuff
in.  Sent it off and it said 'thank you' but added 'we don't reply' so I
have no feedback as to whether it was actioned or what.  I don't know if
developers have a 'fast lane' for such reports, anyway, here's hoping that
if the problem is more wide spread it will quickly attract the appropriate
attention.

I do miss my Chromebook/thonny, but at least I still have my RasPi 400
working nicely.
Btw will your v4 beta install on an Arm based pi or do I need a different
package? - I'd be happy to try and test it if it would be helpful.

aivarannamaa, 2022-04-10:
I now reported it at https://bugs.chromium.org/p/chromium/issues/detail?id=1315000 (The site looks like it's for Chrome, but I've seen people successfully posting Crostini issues there)

> Btw will your v4 beta install on an Arm based pi or do I need a different package? - I'd be happy to try and test it if it would be helpful.

I'm not publishing binaries for Pi, but you can either `pip install thonny` (preferably into a virtual environment), or use https://github.com/thonny/thonny/releases/download/v4.0.0b2/thonny-4.0.0b2.bash which is meant to automatically create a venv, install Thonny and also create a desktop file. (On x86_64 it downloads and installs the binary bundle.)

coloneldare, 2022-04-10:
Thanks - I think your report will be more helpful/focused than mine :-)

coloneldare, 2022-04-22:
Wow! I just downloaded the latest update to ChomeOS (Version 101.0.4951.41
(Official Build) beta (64-bit)),  and I have just started IDLE and Thonny
(3.2.7) and it all seems to be business as usual! Yipee!!    :-)

Hopefully all is now fixed.  But the current Debian/Bullseye offer (3.2.7)
isn't RP2040 (Pico) MicroPython enabled.  Previously I used the Thonny.org
site download with great success, but I'd be happy to jump to your latest
v4 beta and give that a go if helpful, so could you please send me a link
to the latest build and I'll give it a whirl.

Thanks for all your help.  Our thoughts here in the UK are with Ukraine.....

Doug

aivarannamaa, 2022-04-24:
Thanks for the notice!

I can confirm that both IDLE and Thonny work on ChromeOS again.