thonny/thonny #2197: Pico

Womper999, 2022-03-13:
Hi there,

I'm new here and want to program the pico.

I had a problem with the main.py

Now Thonny always gets the message:

Device is busy or does not respond. Your options:

   - wait until it completes current work;
   - use Ctrl+C to interrupt current work;
   - use Stop/Restart to interrupt more and enter REPL.

although no device is connected.
How can I reset Thonny?
I've also reinstalled Thonny to no avail.
I installed version 3.11 and then again 3.3.13.
Unfortunately without success.
What can I do??

aivarannamaa, 2022-03-16:
Thank you for the report!

> although no device is connected.

Do you get this after disconnecting your device or also when you have no device connected and then start Thonny?

Please go to "Run => Select interpreter" and check, which port is configured. Maybe you have another (possibly virtual device) at this port (eg. Bluetooth over serial).

aivarannamaa, 2022-03-23:
Please reopen if you still have this problem and can add more information

closed 2022-03-23 by aivarannamaa